WebODM/WebODM #611: Why the odm_orthophoto .tif change to UTM84-51S

cecabula, 2019-01-29:
### How did you install WebODM?
Docker

### What's your browser and operating system? 
Chrome 71 on Windows 10

### What is the problem?
My coordinate system  is UTM84-51N
Why the odm_orthophoto .tif change to UTM84-51S

### What should be the expected behavior?
I do not know what the setting is changed ?

### How can we reproduce this? 
Upload  7 images of my UAV  & odm_orthophoto .tif 
https://drive.google.com/open?id=1OWqNxj-gjIpVNnBBWH9MGCB4JVF5rYRP
Can you test the 7 images ?
Thanks.

pierotofy, 2019-01-29:
Hey @cecabula this is a bug. We'll have a fix for this shortly.

pierotofy, 2019-01-29:
@cecabula could you try to `./webodm.sh update` and reprocess the dataset? Is the issue solved?

cecabula, 2019-01-29:
./webodm.sh update 
too exciting !
the issue of UTM84-51N is solved.
Thanks.

closed 2019-01-29 by pierotofy