WebODM/WebODM #612: Unable to change the default port of application

deviirmr, 2019-01-29:
Hi,
I have installed a docker instance of WebODM in Ubuntu 16 and it is working fine with default port, but I want to open the application in a different port. I have also edit the ports (to 8001:8000) section in the docker-compose.yml file and restarted the application but it has no impact, unfortunately, it is still active in the default port(8000).
I have followed the post https://github.com/OpenDroneMap/WebODM/issues/235
but nothing works.
Thank You!

pierotofy, 2019-01-29:
Hi @deviirmr,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-02-01 12:55) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

deviirmr, 2019-01-29:
Thank you for the suggestion, next time definitely I will follow the 'Issue Template'
My query is just solved by you in another ticke So I am closing this issue: https://github.com/OpenDroneMap/WebODM/issues/235#issuecomment-458638656

Thank you!

closed 2019-01-29 by deviirmr