WebODM/WebODM #615: [Improvement] UX warn users when disk space is low

pierotofy, 2019-02-03:
An upload will fail with a less useful "186 files cannot be uploaded. As a reminder, only images (.jpg, .png) and GCP files (.txt) can be uploaded. Try again." message, whereas the problem is lack of disk space.

Saijin-Naib, 2021-06-09:
Following on from this warning, would it be possible to have WebODM gracefully abort processing when disk space on the volume it is installed on drops to 10% free?

Something like that would have prevented the user here from filling their disk:  
https://community.opendronemap.org/t/se-llena-disco-al-100-bloquea-linux/7722

pierotofy, 2021-06-09:
It might be possible to add this in NodeODM.

Saijin-Naib, 2021-06-09:
> It might be possible to add this in NodeODM. I don't think it's as straightforward though (and is imperfect, since disk space might be fine during processing, but might fill past 100% when WebODM downloads a copy of the assets).

Yeah, that sounds like a tough thing to manage. Should I open an issue in NodeODM to track this?

pierotofy, 2021-06-09:
I think we can include this in the scope of this existing issue (#615)

CrazyFluffyPony, 2022-11-04:
Worst case scenario is when files can be uploaded, but when processing the files the disk is filled up (which crashes WebODM and prevents a user to delete any files without trying to access the VM manually).

Also, maybe you could add a hint to the installation guide on how to increase the VM Size. Because i spent 3 days trying to increase it until i found you can add `--virtualbox-disk-size "300000"` in `C:\Program Files\Docker Toolbox\start.sh` line 69

20GB default size is good for a test run, but allows no more then ~300 images

_Big thanks to all of you for this nice peace of software!!!_

pierotofy, 2025-03-25:
After thinking for a long time about this, I don't have a good way to implement this. The operating system is ultimately in charge of notifying users of disk space depletion. For WebODM to attempt to do this correctly is a very complex issue and ultimately will not result in a solution that handles all cases. At best, it will catch a small subset. 🙏

closed 2025-03-25 by pierotofy