WebODM/WebODM #616: Unable to start using ./devenv.sh start

NicksonYap, 2019-02-06:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For questions, comments and feature requests, please open a topic on http://community.opendronemap.org/c/webodm. 

Also, is your problem listed here? https://github.com/OpenDroneMap/WebODM#common-troubleshooting
If so, do not open an issue. We already know about it and steps to solve it are already documented.

Every issue needs to follow the format specified below. Do not remove the questions, simply answer each question by typing below it. If you don't know how to answer a question, write so. Questions that don't follow these guidelines will be automatically closed by our friendly IssueBot (who's not very smart and needs to see a particular format to keep an issue open).
****************************************

### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 72 on Linux
Ubuntu 16.04
https://whatismybrowser.com/w/L9HHM9Q

### What is the problem?

Cloned latest repo at 3d3c6164f618c5ee452013f593b8ee9e278b59af

After running successfully: ./devenv.sh start

Error in webpage:

[image]

Console output/steps:
[consoleoutput.txt](https://github.com/OpenDroneMap/WebODM/files/2838451/consoleoutput.txt)

### What should be the expected behavior?

Should be a normal login page

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Just clone new repo, run:

```
./webodm.sh start (works normally)
./webodm.sh stop 
 ./devenv.sh start
```

Produces the  error shown in screenshot above

The following is my SSD partitioning
[image]

I had issue when using exFAT (unable to create symlink)

NicksonYap, 2019-02-06:
I found the way to fix

I was aware this issue is related with the fact that  `envdev.sh` does not work well with `webdom.sh start --media_dir XXXXX`

So when resetting docker I have deleted containers and images, but forgot to delete volumes.

```
docker volume rm webodm_appmedia 
docker volume rm webodm_dbdata
```

So the actual issue is that `envdev.sh` should work with` --media_dir XXXXX `or that
the development should be able to be activated with `webdom.sh start --media_dir XXXXX --dev`

pierotofy, 2019-02-07:
Hey @NicksonYap :hand: thanks for sharing your findings! Admittedly the `devenv.sh` script could probably be merged with `webodm.sh` via a `--dev` flag. I will leave this issue open and leave it as a an improvement.

pierotofy, 2019-02-19:
This is now done (in `febspring` branch).

PR #625 (2019-02-21): February Sprint
Fixes #616
closed 2019-02-19 by pierotofy