WebODM/WebODM #619: Yuneec Typhoon H images fail to load

adcornetto, 2019-02-12:
I posted this to the forum, but since ran a local copy of WebODM and received the below message on the console, which indicated to post it here.  The images are from a Yuneec Typhoon H.

worker | [2019-02-12 03:16:59,086: ERROR/ForkPoolWorker-19] worker.tasks.process_task[486bbb15-bf6f-4e42-ba1b-4e6cd36ea399]: Uncaught error! This is potentially bad. Please report it to http://github.com/OpenDroneMap/WebODM/issues: “dump” got wrong type of exif value.
worker | 41728 in Exif IFD. Got as <class ‘int’>. Traceback (most recent call last):
worker | File “/usr/local/lib/python3.6/site-packages/piexif/_dump.py”, line 272, in _value_to_bytes
worker | value_str = raw_value + b"\x00" * (4 - length)
worker | TypeError: can only concatenate tuple (not “bytes”) to tuple
worker |
worker | During handling of the above exception, another exception occurred:
worker |
worker | Traceback (most recent call last):
worker | File “/usr/local/lib/python3.6/site-packages/piexif/_dump.py”, line 337, in _dict_to_bytes
worker | offset)
worker | File “/usr/local/lib/python3.6/site-packages/piexif/_dump.py”, line 274, in _value_to_bytes
worker | raise ValueError(“Got invalid type to convert.”)
worker | ValueError: Got invalid type to convert.
worker |
worker | During handling of the above exception, another exception occurred:
worker |
worker | Traceback (most recent call last):
worker | File “/webodm/worker/tasks.py”, line 54, in process_task
worker | task.process()

pierotofy, 2019-02-12:
Hi @adcornetto,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-02-14 22:20) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2019-02-12:
In WebODM the images fail to resize with:

```
Traceback (most recent call last):
  File "/webodm/worker/tasks.py", line 54, in process_task
    task.process()
  File "/webodm/app/models/task.py", line 343, in process
    resized_images = self.resize_images()
  File "/webodm/app/models/task.py", line 737, in resize_images
    resized_images = list(map(partial(resize_image, resize_to=self.resize_to, done=callback), images_path))
  File "/webodm/app/models/task.py", line 103, in resize_image
    im.save(resized_image_path, "JPEG", exif=piexif.dump(exif_dict), quality=100)
  File "/usr/local/lib/python3.6/site-packages/piexif/_dump.py", line 68, in dump
    exif_set = _dict_to_bytes(exif_ifd, "Exif", zeroth_length)
  File "/usr/local/lib/python3.6/site-packages/piexif/_dump.py", line 341, in _dict_to_bytes
    '{0} in {1} IFD. Got as {2}.'.format(key, ifd, type(ifd_dict[key]))
ValueError: "dump" got wrong type of exif value.
41728 in Exif IFD. Got as <class 'int'>.
```

pierotofy, 2019-02-12:
Minimal program to reproduce:

```
from PIL import Image
import piexif
i = Image.open("YUN00227.jpg")
d = piexif.load(i.info['exif'])
piexif.dump(d)
```

pierotofy, 2019-02-12:
Actually, it's more of a problem with the YUNEEC, as it's incorrectly encoding the "FileSource" exif field as type "undefined" instead of "int".

This will require some special case handling.

pierotofy, 2019-02-12:
Ok so the problem is with the YUNEEC, the field should be encoded as binary, but it's int instead.

pierotofy, 2019-02-12:
Temporary workaround is to choose `Resize: Yes (In Browser)` when creating a new task since the exif parser we use for in browser resize is more fault tolerant.

pierotofy, 2019-02-12:
Lightning (via NodeODM) should work now. If you try again the images will get accepted and you should not see the "unrecognized image format" error anymore.

adcornetto, 2019-02-18:
Thanks for the help on this.  Is the fix only for NodeODM or should WebODM also work now?

pierotofy, 2019-02-19:
Workaround for WebODM is to choose "Resize: Yes (In Browser)" as a task option. NodeODM should work without problems.

pierotofy, 2019-06-14:
This should now be fixed in WebODM! If the problem is still there, please re-open the issue?

closed 2019-06-14 by pierotofy