WebODM/WebODM #620: [Feature] Allow upload of zip files

Terranigmus, 2019-02-12:
For uploading it would be very convenient to be able to upload compressed files that contain images since bandwidth is more of a time issue than unzipping on a running server. 

It would certainly improve quality of life.

That is all.

pierotofy, 2019-02-12:
Hey @Terranigmus, you mean allowing a URL to a zip file to be specified (such as a zip file hosted on a website), or just a single zip file (like, select a zip file from your computer and upload that to WebODM)?

You wouldn't save much bandwidth by using a zip file from your computer, JPG already has it's own compression.

Terranigmus, 2019-02-12:
Good point, nevertheless I probably would have less hassle with unzipping/zipping data files as they are often shared in compressed containers, for example when downloading them from Google Drive or some other files-sharing sites,  also it would allow me to split uploads into batches without manually selecting a bunch of files in a folder and so on and so on. 
I was wrong on the bandwidth though, didn't think that through :(

olivierdalang, 2019-06-25:
Allowing to transfer directly from url would be great, eg. to transfer from Dropbox.

smathermather, 2019-07-15:
Agreed. Also, this is a duplicate issue: https://github.com/OpenDroneMap/WebODM/issues/89

closed 2019-07-15 by smathermather