WebODM/WebODM #621: Volume calculation results in reactjs (most often with iframes)

ChipwizBen, 2019-02-12:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox / Chrome - results are consistent

### What is the problem?
As per: https://community.opendronemap.org/t/volume-calculation-results-in-reactjs-most-often-with-iframes/1770/5

When trying to calculate volumes in WebODM (iFrame), the popup display returns blank (just says "Measurement" and Center on this area / Delete). In the browser console, the following error is returned:

https://reactjs.org/docs/error-decoder.html?invariant=31&args[]=object%20with%20keys%20%7BreadyState%2C%20getResponseHeader%2C%20getAllResponseHeaders%2C%20setRequestHeader%2C%20overrideMimeType%2C%20statusCode%2C%20abort%2C%20state%2C%20always%2C%20then%2C%20promise%2C%20pipe%2C%20done%2C%20fail%2C%20progress%2C%20complete%2C%20success%2C%20error%2C%20responseText%2C%20responseJSON%2C%20status%2C%20statusText%7D&args[]=

The issue occurs in both latest versions of Firefox and Chrome.

### What should be the expected behavior?

Volume calculations should work the same in both regular and iFrame displays.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Embed a WebODM iFrame into another site. Try to perform a volume calculation.

closed 2019-02-19 by pierotofy

ChipwizBen, 2019-03-30:
This issue has returned in the current version.

pierotofy, 2019-03-30:
Does it happen with all datasets or just some? Does it happen with a small dataset? Maybe this is a different issue.

ChipwizBen, 2019-03-31:
You're right, it is a different issue. It's a timeout. I was trying to measure too much at once (about 1sq km).

pierotofy, 2019-03-31:
It's still something that should be fixed. Could you open a new issue separate from this one so that we may track it (and fix it, hopefully)? :pray:

rzrmayur, 2022-08-10:
the volume window is showing blank when embedding it into iframe