WebODM/WebODM #626: [FEATURE REQUEST] Ubuntu 18.x

sann1432, 2019-02-19:
Create a native server installation package for Ubuntu 18.x

Include configurations of native services via WebODM UI

pierotofy, 2019-02-19:
Hi @sann1432,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-02-22 11:55) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2019-02-22:
This will be done sometimes between now and 2021 which is the end of life for Ubuntu 16.04. We'd love to get help to speed this up.

pierotofy, 2020-11-24:
All ODM software can now run on 20.04. This can be closed :clinking_glasses:

closed 2020-11-24 by pierotofy