WebODM/WebODM #627: [Feature Request] Restore other export methods back into WebODM/ODM

nixxet, 2019-02-19:
Can you please restore other methods of Point Cloud downloads, such as LAS, LAZ, PLY, CSV?

pierotofy, 2019-02-19:
Hi @nixxet,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-02-22 18:10) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2019-02-22:
Hey @nixxet :hand: thanks for reporting this. There are some formats that we are not going to reintroduce (like PLY, which is a terrible format for point clouds), with https://github.com/OpenDroneMap/ODM/pull/953/files we are going to give the ability to generate LAS files (CSV can already be generated by toggling the `--pc-csv` flag).

Popups currently show all possible downloads supported by WebODM, regardless of whether they are available on the selected dataset, which is something we should improve. I've renamed this issue to reflect that.

nixxet, 2019-02-23:
understood and thank you!

ChipwizBen, 2019-02-27:
Related to this, is the any way to _disable_ the popup entirely? I pull downloads over the public file links so the popup is redundant and gets in the way when viewing the map.

pierotofy, 2019-02-27:
You might be able to add some custom Javascript code in the Theme Settings that intercepts the popup from displaying, or use CSS rules to hide it.

ChipwizBen, 2019-02-27:
Ah, top tip, didn't think of that. For future Googlers:
`.leaflet-popup {display: none}`

PR #639 (2019-04-02): Fixes and new features
Fixes #627

Do not merge, more stuff coming.
closed 2019-04-02 by pierotofy