WebODM/WebODM #630: Feature Request - Introduction of image collections

epurnell, 2019-02-22:
When running on a server the resources consumed by repeatedly uploading the images is huge. There is not an easy way of managing them once loaded either. 

**Part implemented in 8.1 thanks Piero!!** 

Further improvement would be to separate the image loading stage from the project processing stage. This introduces the concept of ‘image sets or collection lets say’

Load an image set for processing
Add project and select image set
Process 1 and check result
Process 2 and check result
Repeat until the project is complete using deferent settings or nodes... 

Also to think about as part of this is.... 
Manage image sets plugin? 

API endpoints for custom image loading directly into the app (from website/CMS/file system/ DB such as Postgres/ QGIS) 

Best - Edd 👍🏻😀

pierotofy, 2019-02-22:
Hi @epurnell,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-02-25 04:45) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2021-10-14:
The current workflow is to use the Cloud Import plugin to load external image sets, if one wants to avoid uploading images directly. There's also #939 that could improve the workflow. I'm going to close this.

closed 2021-10-14 by pierotofy