WebODM/WebODM #638: Fixed Problem: Ubuntu 16.04 on Proxmox with Server Installer

Wlanfr3ak, 2019-03-25:
### How did you install WebODM? (Docker, natively, ...)?

Server Installer SCRIPT_VERSION=1.3.4

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome / Win10 64Bit

### What is the problem?

With normal  Ubuntu 16.04 Installation with Server Installer all was okay but when i try to start a Task everytime after the Picture Resizing i get the Error:

File "run.py", line 11, in 
from scripts.odm_app import ODMApp
File "/home/ubuntu/IIRS/OpenDroneMap/scripts/odm_app.py", line 5, in 
from opendm import types
File "/home/ubuntu/IIRS/OpenDroneMap/opendm/types.py", line 5, in 
from opensfm.exif import sensor_string
ImportError: No module named opensfm.exif

### What should be the expected behavior?

Start to Work and stop and cry later ;-)

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I use Proxmox to separate the Software in an easily way. At the Creating from the VM i use the standard settings. After lots of searhing i found out that CPU Types generates Errors in comipling Processes. So i change the CPU to the Host CPU and not an KVM one and run the installer again. Now it works. Please add a Notice for Virtualization Users that the Host CPU Type is the best way to use WebODM. I hope that other User can find here a solution for a similiar Problem.

ichfred, 2019-03-25:
"No module named opensfm.exif"

I have the a similar problem. Bought the Installer and used a fresh installed Ubuntu 16.04.6 i386 Server OS

pierotofy, 2019-03-26:
Thanks for reporting this @Wlanfr3ak !

@ichfred are you running the installer on a 64bit CPU? i386 is a 32bit architecture.

ichfred, 2019-04-01:
> @ichfred are you running the installer on a 64bit CPU? i386 is a 32bit architecture.

Typo from Copy&Paste. Should be Ubuntu 16.04.6 amd64. I fixed it with running the installation script again.

closed 2020-04-02 by pierotofy