WebODM/WebODM #640: Error: Process exited with code 1

afrkiev, 2019-03-29:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For questions, comments and feature requests, please open a topic on http://community.opendronemap.org/c/webodm. 

Also, is your problem listed here? https://github.com/OpenDroneMap/WebODM#common-troubleshooting
If so, do not open an issue. We already know about it and steps to solve it are already documented.

Every issue needs to follow the format specified below. Do not remove the questions, simply answer each question by typing below it. If you don't know how to answer a question, write so. Questions that don't follow these guidelines will be automatically closed by our friendly IssueBot (who's not very smart and needs to see a particular format to keep an issue open).
****************************************

### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 66 on Windows 10 Home 8GB RAM

### What is the problem?

Process exited with code 1

### What should be the expected behavior?

Completed map / object

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Uploaded 8 images for test data from Samsung Galaxy S8.

I've set min-num-features to 12000 or 20000 and it didn't worked.

Resized pics to 640 - no result.

Set mesh-octree-depth: 15

Trying to get the best output to evaluate this tool as an alternative for orthos and 3d models, etc from my drone work.

VM has 5gb memory and 35 ssd space allocated

[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /var/www/data/9a326198-237f-406f-97c5-2a80301afaa7/images
[DEBUG]   /var/www/data/9a326198-237f-406f-97c5-2a80301afaa7/images
[DEBUG]   Loaded 20190329_084736.jpg | camera: samsung sm-g950f | dimensions: 640 x 360 | lat: 47.8661111111 | lon: 35.1055555556 | alt: 88.0
[DEBUG]   Loaded 20190329_084726.jpg | camera: samsung sm-g950f | dimensions: 640 x 360 | lat: 47.8661111111 | lon: 35.1055555556 | alt: 88.0
[DEBUG]   Loaded 20190329_084711.jpg | camera: samsung sm-g950f | dimensions: 640 x 360 | lat: 47.8661111111 | lon: 35.1055555556 | alt: 88.0
[DEBUG]   Loaded 20190329_084719.jpg | camera: samsung sm-g950f | dimensions: 640 x 360 | lat: 47.8661111111 | lon: 35.1055555556 | alt: 88.0
[DEBUG]   Loaded 20190329_084742.jpg | camera: samsung sm-g950f | dimensions: 640 x 360 | lat: 47.8661111111 | lon: 35.1055555556 | alt: 88.0
[DEBUG]   Loaded 20190329_084751.jpg | camera: samsung sm-g950f | dimensions: 640 x 360 | lat: 47.8661111111 | lon: 35.1055555556 | alt: 88.0
[DEBUG]   Loaded 20190329_084705.jpg | camera: samsung sm-g950f | dimensions: 640 x 360 | lat: 47.8661111111 | lon: 35.1055555556 | alt: 88.0
[DEBUG]   Loaded 20190329_084731.jpg | camera: samsung sm-g950f | dimensions: 640 x 360 | lat: 47.8661111111 | lon: 35.1055555556 | alt: 88.0
[INFO]    Wrote images database: /var/www/data/9a326198-237f-406f-97c5-2a80301afaa7/images.json
[INFO]    Found 8 usable images
[DEBUG]   running /code/build/bin/odm_extract_utm -imagesPath /var/www/data/9a326198-237f-406f-97c5-2a80301afaa7/images/ -imageListFile /var/www/data/9a326198-237f-406f-97c5-2a80301afaa7/img_list.txt -outputCoordFile /var/www/data/9a326198-237f-406f-97c5-2a80301afaa7/odm_georeferencing/coords.txt  -logFile /var/www/data/9a326198-237f-406f-97c5-2a80301afaa7/odm_georeferencing/odm_georeferencing_utm_log.txt
Warning:
Directory Photo has an unexpected next pointer; ignored.
Warning:
Directory Iop has an unexpected next pointer; ignored.
Warning:
Directory GPSInfo has an unexpected next pointer; ignored.
Warning:
Directory Photo has an unexpected next pointer; ignored.
Warning:
Directory Iop has an unexpected next pointer; ignored.
Warning:
Directory GPSInfo has an unexpected next pointer; ignored.
Warning:
[... 255 lines cut ...]
File "/usr/local/lib/python2.7/dist-packages/loky/_base.py", line 589, in result_iterator
yield future.result()
File "/usr/local/lib/python2.7/dist-packages/loky/_base.py", line 433, in result
return self.__get_result()
File "/usr/local/lib/python2.7/dist-packages/loky/_base.py", line 381, in __get_result
raise self._exception
Exception: Child returned 1

This was caused directly by
"""
Traceback (most recent call last):
File "/usr/local/lib/python2.7/dist-packages/loky/process_executor.py", line 418, in _process_worker
r = call_item()
File "/usr/local/lib/python2.7/dist-packages/loky/process_executor.py", line 272, in __call__
return self.fn(*self.args, **self.kwargs)
File "/usr/local/lib/python2.7/dist-packages/loky/process_executor.py", line 337, in _process_chunk
return [fn(*args) for args in chunk]
File "/code/opendm/dem/commands.py", line 83, in create_dem
pdal.run_pipeline(json, verbose=verbose)
File "/code/opendm/dem/pdal.py", line 163, in run_pipeline
out = system.run(' '.join(cmd) + ' > /dev/null 2>&1')
File "/code/opendm/system.py", line 34, in run
raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
"""

afrkiev, 2019-03-29:
Sorry for disturbing. I had bad dataset

closed 2019-03-29 by afrkiev

smathermather, 2019-03-29:
Thanks for closing the issue!