WebODM/WebODM #642: Server Installer built servers do not survive reboots

ChipwizBen, 2019-03-31:
### How did you install WebODM? (Docker, natively, ...)?

Native Server Installer

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

N/A

### What is the problem?

After using the installer and rebooting, WebODM/NodeODM does not start.

```
Failed to restart nodeodm.service: Unit nodeodm.service not found.
Failed to restart webodm-nginx.service: Unit webodm-nginx.service not found.
Failed to restart webodm-gunicorn.service: Unit webodm-gunicorn.service not found.
Failed to restart webodm-celery.service: Unit webodm-celery.service not found.
Failed to restart webodm-celerybeat.service: Unit webodm-celerybeat.service not found.
```

```
root@proc01:~# systemctl list-unit-files | grep odm
nodeodm.service                            enabled 
webodm-celery.service                      enabled 
webodm-celerybeat.service                  enabled 
webodm-gunicorn.service                    enabled 
webodm-nginx.service                       enabled 
```

I also ran the upgrade process, but that did not fix the issue. I've had this issue on 2/2 servers that I've tried it on.
The only exceptional thing about these servers is that they're running Ceph mounts (kernel/fstab) for the directories. Other than that, they are clean installs of 16.04:
```
root@proc01:~# df -h
Filesystem                                                             Size  Used Avail Use% Mounted on
udev                                                                    69G     0   69G   0% /dev
tmpfs                                                                   14G  8.7M   14G   1% /run
/dev/mapper/proc01--vg-root                                             14G  4.5G  8.9G  34% /
tmpfs                                                                   69G  4.0K   69G   1% /dev/shm
tmpfs                                                                  5.0M     0  5.0M   0% /run/lock
tmpfs                                                                   69G     0   69G   0% /sys/fs/cgroup
/dev/sda1                                                              720M  107M  577M  16% /boot
10.0.0.5:6789,10.0.0.6:6789,10.0.0.7:6789:/NFS/proc01/www      22T  3.4T   19T  16% /www
10.0.0.5:6789,10.0.0.6:6789,10.0.0.7:6789:/NFS/proc01/code     22T  3.4T   19T  16% /code
10.0.0.5:6789,10.0.0.6:6789,10.0.0.7:6789:/NFS/proc01/webodm   22T  3.4T   19T  16% /webodm
tmpfs                                                                   14G     0   14G   0% /run/user/0

```

Essentially, the installer works fine and the services start on the first try, but as soon as the system is restarted it does not recover.

### What should be the expected behavior?

The services should survive reboots.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Install from Server Installer. Reboot.

ChipwizBen, 2019-03-31:
OK, I rebuilt nodes 1 and 2 again from completely clean disks. This is definitely an issue. Here is node 1 after a reboot:
```
root@proc01:~# ps aux | grep odm
root      2621  0.0  0.0  14224   936 pts/0    S+   18:54   0:00 grep --color=auto odm
root@proc01:~# service webodm status
● webodm.service
   Loaded: not-found (Reason: No such file or directory)
   Active: inactive (dead)
root@proc01:~# systemctl list-unit-files | grep odm
nodeodm.service                            enabled 
webodm-celery.service                      enabled 
webodm-celerybeat.service                  enabled 
webodm-gunicorn.service                    enabled 
webodm-nginx.service                       enabled 
root@proc01:~# service nodeodm status
● nodeodm.service
   Loaded: not-found (Reason: No such file or directory)
   Active: inactive (dead)
root@proc01:~# 
```

And here is node 2 that hasn't been rebooted yet:
```
root@proc02:~# ps aux | grep odm
odm      26480  0.2  0.0 1090752 55820 ?       Ssl  18:39   0:01 /usr/bin/node index.js
root     26528  0.0  0.0 125116  1460 ?        Ss   18:39   0:00 nginx: master process /usr/sbin/nginx -c /webodm/nginx/nginx.conf
odm      26529  0.0  0.0 125548  3132 ?        S    18:39   0:00 nginx: worker process
odm      26560  0.1  0.0 489688 71372 ?        Ss   18:39   0:01 /webodm/python3-venv/bin/python3 /webodm/python3-venv/bin/gunicorn webodm.wsgi --bind unix:/webodm/gunicorn.sock --timeout 300000 --max-requests 250 --preload
odm      26592  0.4  0.0 507156 84452 ?        Ss   18:39   0:03 /webodm/python3-venv/bin/python3 /webodm/python3-venv/bin/celery -A worker worker --autoscale 8,2 --max-tasks-per-child 1000 --loglevel=warn
odm      26625  0.3  0.0 928632 107360 ?       Ss   18:39   0:02 /webodm/python3-venv/bin/python3 /webodm/python3-venv/bin/celery -A worker beat --pidfile=
odm      26651  0.0  0.0 498136 55884 ?        S    18:40   0:00 /webodm/python3-venv/bin/python3 /webodm/python3-venv/bin/gunicorn webodm.wsgi --bind unix:/webodm/gunicorn.sock --timeout 300000 --max-requests 250 --preload
postgres 26653  0.0  0.0 306200 16020 ?        Ss   18:40   0:00 postgres: 9.6/main: postgres webodm_dev ::1(34418) idle
odm      26874  0.1  0.0 507668 65504 ?        S    18:45   0:00 /webodm/python3-venv/bin/python3 /webodm/python3-venv/bin/celery -A worker worker --autoscale 8,2 --max-tasks-per-child 1000 --loglevel=warn
odm      26915  0.0  0.0 507412 65224 ?        S    18:46   0:00 /webodm/python3-venv/bin/python3 /webodm/python3-venv/bin/celery -A worker worker --autoscale 8,2 --max-tasks-per-child 1000 --loglevel=warn
root     27228  0.0  0.0  14224  1088 pts/0    S+   18:53   0:00 grep --color=auto odm
```

Both had identical build procedures.

ChipwizBen, 2019-03-31:
OK, I found the problem, and a small change to the script can save a lot of headaches. The service files are stored in /www/service/ and /webodm/service/, and these are referenced from symlinks:

```
root@proc01:~# ll /etc/systemd/system | grep odm
lrwxrwxrwx 1 root root   29 Mar 31 18:39 nodeodm.service -> /www/services/nodeodm.service
lrwxrwxrwx 1 root root   41 Mar 31 18:39 webodm-celerybeat.service -> /webodm/service/webodm-celerybeat.service
lrwxrwxrwx 1 root root   37 Mar 31 18:39 webodm-celery.service -> /webodm/service/webodm-celery.service
lrwxrwxrwx 1 root root   39 Mar 31 18:39 webodm-gunicorn.service -> /webodm/service/webodm-gunicorn.service
lrwxrwxrwx 1 root root   36 Mar 31 18:39 webodm-nginx.service -> /webodm/service/webodm-nginx.service
```
This is annoying, because things like `After=local-fs.target remote-fs.target` are useless because _the service file is not readable to action this until the directory is readable_ as the mount has not yet completed. This is a chicken and egg situation. The service files should be stored in /etc/systemd/system/, not just as symlinks.

I moved all the service files to /etc/systemd/system/, added `After=local-fs.target remote-fs.target`, and now it works perfectly. Can I suggest that the service files are added to that location by default, and also depend on all mounts completing (local and remote)?

pierotofy, 2020-04-02:
Thanks for documenting the steps @ChipwizBen ! I'll close this for now as it relates to the installer (which needs fixing on my part, not necessarily in WebODM).

closed 2020-04-02 by pierotofy