WebODM/WebODM #644: 403 Forbidden trying to use contours in iframe

ChipwizBen, 2019-04-03:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Linux / Firefox

### What is the problem?

When trying to use the contour feature from within an iframe it results in a 403 Forbidden browser log message and the following on-screen message: 

> Cannot retrieve information for task cfb2de00-06bc-4b03-8348-ad5afa43a12c. Are you are connected to the internet.

### What should be the expected behavior?

The contour interface should display correctly.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Try to load contours from within an iframe.

pierotofy, 2019-04-03:
Thanks for reporting this issue! Definitely a bug.

closed 2019-04-03 by pierotofy

pierotofy, 2019-04-03:
See if it's solved now @ChipwizBen?

PR #645 (2019-04-03): Contours fix in iframes
Fixes #644 as well as popup assets list error.