WebODM/WebODM #647: I can't find this “posm-gcpi ” page.

MH-Blog, 2019-04-04:
hello,I want to know where this page is located in the engineering file, and I want to modify some parameters in it.Thank you!

pierotofy, 2019-04-04:
Hi @MH-Blog,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-04-07 04:15) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

smathermather, 2019-04-04:
@MH-Blog -- the bot is right. This is a question better posted at https://community.opendronemap.org

closed 2019-04-04 by pierotofy