WebODM/WebODM #648: Hyphens in filenames break opensfm

ChipwizBen, 2019-04-04:
### How did you install WebODM? (Docker, natively, ...)?

Docker + Native

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

N/A

### What is the problem?

Hyphens in filenames break opensfm
https://community.opendronemap.org/t/fault-with-gcp-list-txt-processing/

### What should be the expected behavior?

Filenames should be POSIX compatible.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

...

smathermather, 2019-04-04:
Agree this is a bug based on what I see in the community post. But if this is an OpenSfM issue, it might be most effective to also submit an issue there: https://github.com/mapillary/OpenSfM/issues

pierotofy, 2019-04-04:
Yep, this should be fixed in OpenSfM. We can track it here as well just for record.

ChipwizBen, 2019-04-04:
Feel free to add to: https://github.com/mapillary/OpenSfM/issues/412

pierotofy, 2019-04-10:
Actually, upon further inspection, I think this is caused by NodeODM's sanitization of filenames. So it's not OpenSfM's fault (the issue should be closed there).

I've pushed a change in NodeODM that allows dashes/hyphens as a temporary workaround,  although a better fix will require sanitizing the GCP list as well. https://github.com/OpenDroneMap/NodeODM/issues/77

pierotofy, 2020-04-02:
This should be fixed. If not, please re-open. :clinking_glasses:

closed 2020-04-02 by pierotofy