WebODM/WebODM #652: Nginx package name has changed

pierotofy, 2019-04-16:
It's no longer `nginx`, but `nginx-core`.

This should have broken the docker images and should be checked.

pierotofy, 2019-04-16:
Hi @pierotofy,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-04-19 17:50) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

smathermather, 2019-04-16:
This is very meta...
[image]

pierotofy, 2019-04-18:
lol, forgot to follow my own rules.

pierotofy, 2020-04-02:
This is fixed.

closed 2020-04-02 by pierotofy