WebODM/WebODM #653: Contours file could not be generated. This might be a bug

enrigarcar, 2019-04-20:
Hi, I’ve been working with WebODM since few months ago (Congratulations!), and when contours tool were launched, I quickly updated it. Buy I realized that doesn’t work fine when --media-dir parameter is applied.
Here is my Google Drive link for the error.
https://drive.google.com/file/d/1pwtacpRtzq9MWUOasm7C16IbRUAX-9p2/view?usp=sharing 1

Thanks in advantage!

pierotofy, 2019-04-20:
Hi @enrigarcar,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-04-22 20:55) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

enrigarcar, 2019-04-20:
Sorry! It's my first Issue.

How did you install WebODM? (Docker, natively, ...)?
Docker

What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Firefox 60 on Linux (but i've been working with Chrome 73 on Linux)

What is the problem?
WebODM contours doesn’t work fine when --media-dir parameter is applied.

What should be the expected behavior?
Contours lines displayed on ortophoto

How can we reproduce this?
Restart WebODM with --media-dir param.
Example: ./WebODM/webodm.sh restart --media-dir /media/enrique/Storage/WebODM/
Here the error message:
https://drive.google.com/file/d/1pwtacpRtzq9MWUOasm7C16IbRUAX-9p2/view?usp=sharing

pierotofy, 2020-04-02:
Hey @enrigarcar :hand: we've made some big changes to the GRASS engine lately, is this error still happening?

pierotofy, 2020-11-24:
Closing and assuming to be fixed.

closed 2020-11-24 by pierotofy