WebODM/WebODM #654: Texturing exited with code 1 - Segmentation Fault

orcunbaslak, 2019-04-21:
Updated the software via ./webodm.sh update and running the latest version as of now.

Here's my config.

Created on: 4/21/2019, 9:43:58 PM
Processing Node: node-odm-1:3000 (auto)
Options: min-num-features: 8000, rerun-from: mvs_texturing, orthophoto-resolution: 200, dem-resolution: 5, opensfm-depthmap-min-consistent-views: 4, use-exif: true, dem-decimation: 80, dsm: true, texturing-keep-unseen-faces: true, pc-csv: true

Just 5 aerial photos.

`[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /var/www/data/af12567f-1b6e-4ba8-848b-dd01269800c1/images
[INFO]    Loading images database: /var/www/data/af12567f-1b6e-4ba8-848b-dd01269800c1/images.json
[INFO]    Found 5 usable images
[ERROR]   Failed parsing GPS position for vlcsnap-2019-04-21-21h28m51s921.png, skipping
[ERROR]   Failed parsing GPS position for vlcsnap-2019-04-21-21h28m47s448.png, skipping
[ERROR]   Failed parsing GPS position for vlcsnap-2019-04-21-21h28m36s235.png, skipping
[ERROR]   Failed parsing GPS position for vlcsnap-2019-04-21-21h28m57s201.png, skipping
[ERROR]   Failed parsing GPS position for vlcsnap-2019-04-21-21h28m41s670.png, skipping
[WARNING] Could not generate coordinates file. Ignore if there is a GCP file
[WARNING] Could not find file /var/www/data/af12567f-1b6e-4ba8-848b-dd01269800c1/odm_georeferencing/coords.txt
[INFO]    Running ODM Load Dataset Cell - Finished
[INFO]    Running ODM OpenSfM Cell
[WARNING] Found a valid OpenSfM reconstruction file in: /var/www/data/af12567f-1b6e-4ba8-848b-dd01269800c1/opensfm/reconstruction.json
[WARNING] Found a valid Bundler file in: /var/www/data/af12567f-1b6e-4ba8-848b-dd01269800c1/opensfm/reconstruction.json
[INFO]    Running ODM OpenSfM Cell - Finished
[INFO]    Running SMVS Cell
[WARNING] Found a valid SMVS reconstruction file in: /var/www/data/af12567f-1b6e-4ba8-848b-dd01269800c1/smvs/smvs_dense_point_cloud.ply
[INFO]    Running ODM SMVS Cell - Finished
[INFO]    Running ODM Meshing Cell
[WARNING] Found a valid ODM Mesh file in: /var/www/data/af12567f-1b6e-4ba8-848b-dd01269800c1/odm_meshing/odm_mesh.ply
[WARNING] Found a valid ODM 2.5D Mesh file in: /var/www/data/af12567f-1b6e-4ba8-848b-dd01269800c1/odm_meshing/odm_25dmesh.ply
[INFO]    Running ODM Meshing Cell - Finished
[INFO]    Running MVS Texturing Cell
[DEBUG]   Writing MVS Textured file in: /var/www/data/af12567f-1b6e-4ba8-848b-dd01269800c1/odm_texturing/odm_textured_model.obj
[DEBUG]   running /code/SuperBuild/install/bin/texrecon /var/www/data/af12567f-1b6e-4ba8-848b-dd01269800c1/opensfm/reconstruction.nvm /var/www/data/af12567f-1b6e-4ba8-848b-dd01269800c1/odm_meshing/odm_mesh.ply /var/www/data/af12567f-1b6e-4ba8-848b-dd01269800c1/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none     --keep_unseen_faces  -n 65535
/code/SuperBuild/install/bin/texrecon (built on Apr 10 2019, 01:38:10)
Load and prepare mesh:
PLY Loader: comment VTK generated PLY File
Reading PLY: 24603 verts...
47620 faces...
done.
Warning: Zero-length normals detected: 0 face normals, 714 vertex normals
Generating texture views:
NVM: Loading file...
NVM: Number of views: 4
NVM: Number of features: 0
Loading 100%... done. (Took 0.312s)
Building adjacency graph:
Adding edges 100%... done. (Took 0.081s)
71352 total edges.
View selection:
Building BVH from 47620 faces...
done. (Took: 58 ms)
Calculating face qualities 100%... done. (Took 0.314s)
Postprocessing face infos 100%... done. (Took 0.017s)
Maximum quality of a face within an image: 834.456
Clamping qualities to 163.82 within normalization.
Writing data cost file...
done.
Optimizing:
Time[s]	Energy
0	48755
0	48663
0	48082
0	47799
0	47657
0	47579
0	47527
0	47506
0	47476
0	47460
0	47452
0	47436
0	47428
0	47424
2610 faces have not been seen
Took: 1.079s
Generating texture patches:
Running...
done. (Took 0.129s)
240 texture patches.
Running global seam leveling:
Create matrices for optimization...
done.
Lhs dimensionality: 25198 x 25198
Calculating adjustments:
Color channel 2: CG took 79 iterations. Residual is 8.7837e-05
Color channel 1: CG took 85 iterations. Residual is 9.70517e-05
Color channel 0: CG took 89 iterations. Residual is 7.91853e-05
Took 0.032 seconds
Adjusting texture patches 100%... done. (Took 0.093s)
Running local seam leveling:
Blending texture patches 100%... done. (Took 1.463s)
Generating texture atlases:
Sorting texture patches...
done.
Working on atlas 1 100%... done.
Finalizing texture atlases...
done. (Took: 2.19s)
Building objmodel:
Saving model...
done.
Whole texturing procedure took: 6.782s
[DEBUG]   Writing MVS Textured file in: /var/www/data/af12567f-1b6e-4ba8-848b-dd01269800c1/odm_texturing_25d/odm_textured_model.obj
[INFO]    Removing old tmp directory /var/www/data/af12567f-1b6e-4ba8-848b-dd01269800c1/odm_texturing_25d/tmp
[DEBUG]   running /code/SuperBuild/install/bin/texrecon /var/www/data/af12567f-1b6e-4ba8-848b-dd01269800c1/opensfm/reconstruction.nvm /var/www/data/af12567f-1b6e-4ba8-848b-dd01269800c1/odm_meshing/odm_25dmesh.ply /var/www/data/af12567f-1b6e-4ba8-848b-dd01269800c1/odm_texturing_25d/odm_textured_model -d gmi -o gauss_clamping -t none     --keep_unseen_faces --nadir_mode -n 65535
/code/SuperBuild/install/bin/texrecon (built on Apr 10 2019, 01:38:10)
Load and prepare mesh:
Reading PLY: 4 verts... 2 faces... done.
Generating texture views:
NVM: Loading file...
NVM: Number of views: 4
NVM: Number of features: 0
Loading 100%... done. (Took 0.336s)
Building adjacency graph:
Adding edges 100%... done. (Took 0s)
1 total edges.
View selection:
Building BVH from 2 faces...
done. (Took: 0 ms)
Calculating face qualities 100%... done. (Took 0.162s)
Postprocessing face infos 100%... done. (Took 0s)
Maximum quality of a face within an image: 0
Clamping qualities to 0 within normalization.
Writing data cost file...
done.
Optimizing:
Time[s]	Energy
0	2
0	2
0	2
0	2
0	2
0	2
0	2
2 faces have not been seen
Took: 0.164s
Generating texture patches:
Running...
done. (Took 0.091s)
0 texture patches.
Running global seam leveling:
Create matrices for optimization...
done.
Lhs dimensionality: 0 x 0
Calculating adjustments:
Received signal SIGSEGV (segmentation fault)
Received signal SIGSEGV (segmentation fault)
Received signal SIGSEGV (segmentation fault)Obtained
8 stack frames: 0x5a5f14 0x7f496cfc84b0Obtained
6 stack frames:
0x4d76a6 0x7f496d79bcbf 0x4d9f19 0x44f6cf 0x7f496cfb3830
0x452d79
0x5a5f14 0x7f496cfc84b0 0x4d76a6
0x7f496d79f43e 0x7f496d3646ba 0x7f496d09a41d
/code/SuperBuild/install/bin/texrecon(_ZN4util6system17print_stack_traceEv+0x34)[0x5a5f14]
/code/SuperBuild/install/bin/texrecon(_ZN4util6system17print_stack_traceEv+0x34)[0x5a5f14]
/lib/x86_64-linux-gnu/libc.so.6(+0x354b0)[0x7f496cfc84b0]
/code/SuperBuild/install/bin/texrecon[0x4d76a6]

/usr/lib/x86_64-linux-gnu/libgomp.so.1(+0xf43e)[0x7f496d79f43e]
Obtained 6 stack frames:/lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba)[0x7f496d3646ba]
0x5a5f14 0x7f496cfc84b0 0x4d76a6 0x7f496d79f43e 0x7f496d3646ba 0x7f496d09a41d
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7f496d09a41d]
Segmentation fault
/code/SuperBuild/install/bin/texrecon(_ZN4util6system17print_stack_traceEv+0x34)[0x5a5f14]
Traceback (most recent call last):
File "/code/run.py", line 47, in <module>
plasm.execute(niter=1)
File "/code/scripts/mvstex.py", line 126, in process
'-n {nadirWeight}'.format(**kwargs))
File "/code/opendm/system.py", line 34, in run
raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1`

pierotofy, 2019-04-21:
Hi @orcunbaslak,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-04-24 15:00) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

orcunbaslak, 2019-04-21:
Hello @pierotofy ,
Just updated according to the template.

pierotofy, 2020-12-07:
It's been a while since this was opened, and we made some major improvements to ODM. Could you try to reprocess with the latest version and see if the issue persists? If so, feel free to reopen this one. :+1:

closed 2020-12-07 by pierotofy