WebODM/WebODM #656: Native install chown could be better

ChipwizBen, 2019-04-24:
### How did you install WebODM? (Docker, natively, ...)?

Native script

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

N/A

### What is the problem?

The update process takes an exceptionally long time due to the chowning of files on line 72 and line 92.
```
sudo chown $(whoami) -R /www /code /webodm
[..]
sudo chown -R odm:odm /code /www /webodm /home/odm
```

### What should be the expected behavior?

Is the first chown really necessary? And could the second chown be removed by running the other commands (such as git pull) with sudo? e.g.: `sudo -u odm git pull origin master`

It seems that chowning the files in this way is the enemy of scale.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Have a lot of files. Update native install.

pierotofy, 2019-04-24:
So the command was added there because we had issues running node (`npm install` in particular). So as far as "is it necessary", the answer is yes. I don't think it's the best solution, but I don't know of a workaround at the moment.

ChipwizBen, 2019-04-24:
Could the chown scope be narrowed to exclude the photo/result files/directories?

pierotofy, 2020-04-02:
Took note time ago, so this can be closed. Thanks!

closed 2020-04-02 by pierotofy