WebODM/WebODM #657: Deleting tasks during upload causes them to get stuck in a 'Deleting' state

ChipwizBen, 2019-04-26:
### How did you install WebODM? (Docker, natively, ...)?

Native

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Any

### What is the problem?

When a task is uploading and you hit "Delete", it locks in this state.

### What should be the expected behavior?

Task should be deleted.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Start task, wait for "Uploading to processing node". Delete task.

ChipwizBen, 2019-06-05:
This is still an issue, but I've also found that tasks get stuck in a deleting state when deleted part-processed, not just during upload.

pierotofy, 2019-06-08:
Any logs? Error messages?

ChipwizBen, 2019-06-10:
Nothing useful. I made a fake job [fd] to test the cancel/delete but it worked fine. I then tried cancelling a real job (that was stuck/frozen) at 13:45, but nothing showed for it in the logs when deleted. It's not clear when it froze or why, but it definitely didn't run out of resources. I've included some logs from earlier in the night - perhaps they might be useful. WebODM and NodeODM are on different physical servers, but I'm not using CluserODM.

Console:
```
INDEXING: 52,000,000 points processed; 52,000,000 points written; 227.94 seconds passed

INDEXING: 55,000,000 points processed; 55,000,000 points written; 236.618 seconds passed

INDEXING: 57,000,000 points processed; 57,000,000 points written; 241.38 seconds passed

INDEXING: 58,000,000 points processed; 58,000,000 points written; 243.698 seconds passed
```

Logs:
```
Jun 10 03:14:44 survey celery[1774]: [2019-06-10 03:14:44,614: ERROR/MainProcess] Task handler raised error: WorkerLostError('Worker exited prematurely: signal 15 (SIGTERM).',)
Jun 10 03:14:44 survey celery[1774]: Traceback (most recent call last):
Jun 10 03:14:44 survey celery[1774]:   File "/webodm/python3-venv/lib/python3.5/site-packages/billiard/pool.py", line 1223, in mark_as_worker_lost
Jun 10 03:14:44 survey celery[1774]:     human_status(exitcode)),
Jun 10 03:14:44 survey celery[1774]: billiard.exceptions.WorkerLostError: Worker exited prematurely: signal 15 (SIGTERM).
Jun 10 03:47:35 survey celery[1774]: [2019-06-10 03:47:35,267: ERROR/ForkPoolWorker-3555] Task worker.tasks.process_task[2bf435e7-64e7-40b8-985f-e952f49e4ba5] raised unexpected: UnboundLocalError("local variable 'have_lock' referenced before assignment",)
Jun 10 03:47:35 survey celery[1774]: Traceback (most recent call last):
Jun 10 03:47:35 survey celery[1774]:   File "/webodm/worker/tasks.py", line 65, in process_task
Jun 10 03:47:35 survey celery[1774]:     have_lock = lock.acquire(blocking=False)
Jun 10 03:47:35 survey celery[1774]:   File "/webodm/python3-venv/lib/python3.5/site-packages/redis/lock.py", line 111, in acquire
Jun 10 03:47:35 survey celery[1774]:     if self.do_acquire(token):
Jun 10 03:47:35 survey celery[1774]:   File "/webodm/python3-venv/lib/python3.5/site-packages/redis/lock.py", line 258, in do_acquire
Jun 10 03:47:35 survey celery[1774]:     client=self.redis))
Jun 10 03:47:35 survey celery[1774]:   File "/webodm/python3-venv/lib/python3.5/site-packages/redis/client.py", line 2944, in __call__
Jun 10 03:47:35 survey celery[1774]:     return client.evalsha(self.sha, len(keys), *args)
Jun 10 03:47:35 survey celery[1774]:   File "/webodm/python3-venv/lib/python3.5/site-packages/redis/client.py", line 2079, in evalsha
Jun 10 03:47:35 survey celery[1774]:     return self.execute_command('EVALSHA', sha, numkeys, *keys_and_args)
Jun 10 03:47:35 survey celery[1774]:   File "/webodm/python3-venv/lib/python3.5/site-packages/redis/client.py", line 668, in execute_command
Jun 10 03:47:35 survey celery[1774]:     return self.parse_response(connection, command_name, **options)
Jun 10 03:47:35 survey celery[1774]:   File "/webodm/python3-venv/lib/python3.5/site-packages/redis/client.py", line 680, in parse_response
Jun 10 03:47:35 survey celery[1774]:     response = connection.read_response()
Jun 10 03:47:35 survey celery[1774]:   File "/webodm/python3-venv/lib/python3.5/site-packages/redis/connection.py", line 624, in read_response
Jun 10 03:47:35 survey celery[1774]:     response = self._parser.read_response()
Jun 10 03:47:35 survey celery[1774]:   File "/webodm/python3-venv/lib/python3.5/site-packages/redis/connection.py", line 284, in read_response
Jun 10 03:47:35 survey celery[1774]:     response = self._buffer.readline()
Jun 10 03:47:35 survey celery[1774]:   File "/webodm/python3-venv/lib/python3.5/site-packages/redis/connection.py", line 216, in readline
Jun 10 03:47:35 survey celery[1774]:     self._read_from_socket()
Jun 10 03:47:35 survey celery[1774]:   File "/webodm/python3-venv/lib/python3.5/site-packages/redis/connection.py", line 174, in _read_from_socket
Jun 10 03:47:35 survey celery[1774]:     data = recv(self._sock, socket_read_size)
Jun 10 03:47:35 survey celery[1774]:   File "/webodm/python3-venv/lib/python3.5/site-packages/redis/_compat.py", line 79, in recv
Jun 10 03:47:35 survey celery[1774]:     return sock.recv(*args, **kwargs)
Jun 10 03:47:35 survey celery[1774]:   File "/webodm/python3-venv/lib/python3.5/site-packages/billiard/common.py", line 125, in _shutdown_cleanup
Jun 10 03:47:35 survey celery[1774]:     sys.exit(-(256 - signum))
Jun 10 03:47:35 survey celery[1774]:   File "/webodm/python3-venv/lib/python3.5/site-packages/billiard/pool.py", line 280, in exit
Jun 10 03:47:35 survey celery[1774]:     return _exit()
Jun 10 03:47:35 survey celery[1774]: SystemExit
Jun 10 03:47:35 survey celery[1774]: During handling of the above exception, another exception occurred:
Jun 10 03:47:35 survey celery[1774]: Traceback (most recent call last):
Jun 10 03:47:35 survey celery[1774]:   File "/webodm/python3-venv/lib/python3.5/site-packages/celery/app/trace.py", line 374, in trace_task
Jun 10 03:47:35 survey celery[1774]:     R = retval = fun(*args, **kwargs)
Jun 10 03:47:35 survey celery[1774]:   File "/webodm/python3-venv/lib/python3.5/site-packages/celery/app/trace.py", line 629, in __protected_call__
Jun 10 03:47:35 survey celery[1774]:     return self.run(*args, **kwargs)
Jun 10 03:47:35 survey celery[1774]:   File "/webodm/worker/tasks.py", line 85, in process_task
[... cut ...]
Jun 09 21:15:04 survey celery[1774]: INFO Automatically assigned processing node proc08.nwk1.com to Task [ECCO] (88b35943-6adb-4af7-a1a9-ae68fb35f063)
Jun 09 21:15:04 survey celery[1774]: [2019-06-09 21:15:04,307: INFO/ForkPoolWorker-1278] Automatically assigned processing node proc08.nwk1.com to Task [ECCO] (88b35943-6adb-4af7-a1a9-ae68fb35f063)
Jun 09 21:15:04 survey celery[1774]: INFO Processing... Task [ECCO] (88b35943-6adb-4af7-a1a9-ae68fb35f063)
Jun 09 21:15:04 survey celery[1774]: [2019-06-09 21:15:04,318: INFO/ForkPoolWorker-1278] Processing... Task [ECCO] (88b35943-6adb-4af7-a1a9-ae68fb35f063)
root@survey:~#
```

fritzagr, 2019-06-25:
i am having this issue when canceling a task during upload to processing node. i am going to try and reproduce the problem on a clean install so the log is not cluttered with other running tasks

fritzagr, 2019-06-25:
ok. i have found a pattern. the cancel only hangs when uploading to a processing node other than localhost. in my case it hangs when uploading to the lightning net or another server on the same network.

fritzagr, 2019-06-25:
below is the only information in the webodm-celery log on the webODM server.

Jun 25 16:27:08 StineDM02 celery[15891]: INFO Processing... Task [Task of 2019-06-25T21:18:58.233Z] (c3f632de-c979-4b97-84e0-6a174fd7f772)
Jun 25 16:27:08 StineDM02 celery[15891]: [2019-06-25 16:27:08,191: INFO/ForkPoolWorker-19711] worker.tasks.process_task[63df9395-3e45-4658-bf89-4cd797f8048c]: Processing... Task [Task of 2019-06-25T21:18:58.233Z] (c3f632de-c979-4b97-84e0-6a174fd7f772)

pierotofy, 2019-06-26:
Thanks @fritzagr. I will try to reproduce this and see if I can find a fix.

pierotofy, 2019-06-26:
Can confirm, this is a bug.

pierotofy, 2019-06-26:
This is fixed in the `cameracalib` branch. Will be available as an update as soon as it's merged into master.

commit b86847bb (2019-06-27): Fixed #657, better cancel task support
closed 2019-06-26 by pierotofy