WebODM/WebODM #659: Task hung at Post Processing after unzip failure

djfurie, 2019-04-30:
### How did you install WebODM? (Docker, natively, ...)?

Docker - Running 0.9 version from dockerhub

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

whatismybrowser.com/w/HXCT45W

### What is the problem?

After running a job using the Lightning network, all.zip failed and the task in WebODM shows as hung in the Post Processing stage.

### What should be the expected behavior?

Tasks completes and results are available through WebODM

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

The lightning job results are here:
https://wln1.nyc3.digitaloceanspaces.com/5ef0c06e-2a51-4cc1-9186-b5254eee9e7c/all.zip

### Some additional questions and comments:

I've manually downloaded all.zip (from the Lightning web site) and successfully extracted it on my workstation (Ubuntu 16.4).

Am I able to load these results into WebODM without having to reprocess the entire task?  I restarted all of the docker containers and my task is now in the Canceled state, however, if Lightning is selected as the processing node, I have no options to restart from a given stage.

Here's the log snippet for when the extraction failed:
`
[2019-04-30 00:00:02,588: WARNING/ForkPoolWorker-525] Task [4/27/19 - Combined] (876e9ede-7233-4516-bd4f-53ccbf6b8b3f) connection/timeout error: HTTPConnectionPool(host='spark1.webodm.net', port=80): Max retries exceeded with url: /task/5ef0c06e-2a51-4cc1-9186-b5254eee9e7c/info?token=c4cdae80-82ac-4613-a002-6f5227da88c3 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fd477969e10>: Failed to establish a new connection: [Errno 111] Connection refused',)). We'll try reprocessing at the next tick.
INFO Processing status: 40 for Task [4/27/19 - Combined] (876e9ede-7233-4516-bd4f-53ccbf6b8b3f)
[2019-04-30 09:58:13,386: INFO/ForkPoolWorker-1129] Processing status: 40 for Task [4/27/19 - Combined] (876e9ede-7233-4516-bd4f-53ccbf6b8b3f)
INFO Downloading all.zip for Task [4/27/19 - Combined] (876e9ede-7233-4516-bd4f-53ccbf6b8b3f)
[2019-04-30 09:58:13,387: INFO/ForkPoolWorker-1129] Downloading all.zip for Task [4/27/19 - Combined] (876e9ede-7233-4516-bd4f-53ccbf6b8b3f)
INFO Extracting all.zip for Task [4/27/19 - Combined] (876e9ede-7233-4516-bd4f-53ccbf6b8b3f)
[2019-04-30 10:07:20,288: INFO/ForkPoolWorker-1129] Extracting all.zip for Task [4/27/19 - Combined] (876e9ede-7233-4516-bd4f-53ccbf6b8b3f)
[2019-04-30 10:07:23,757: ERROR/ForkPoolWorker-1129] worker.tasks.process_task[27d1b1bd-4ae4-442e-b315-84dfc96f007a]: Uncaught error! This is potentially bad. Please report it to http://github.com/OpenDroneMap/WebODM/issues: Error -3 while decompressing data: invalid block type Traceback (most recent call last):
  File "/webodm/worker/tasks.py", line 77, in process_task
    task.process()
  File "/webodm/app/models/task.py", line 632, in process
    self.extract_assets_and_complete()
  File "/webodm/app/models/task.py", line 659, in extract_assets_and_complete
    zip_h.extractall(assets_dir)
  File "/usr/local/lib/python3.6/zipfile.py", line 1501, in extractall
    self._extract_member(zipinfo, path, pwd)
  File "/usr/local/lib/python3.6/zipfile.py", line 1556, in _extract_member
    shutil.copyfileobj(source, target)
  File "/usr/local/lib/python3.6/shutil.py", line 79, in copyfileobj
    buf = fsrc.read(length)
  File "/usr/local/lib/python3.6/zipfile.py", line 849, in read
    data = self._read1(n)
  File "/usr/local/lib/python3.6/zipfile.py", line 925, in _read1
    data = self._decompressor.decompress(data, n)
zlib.error: Error -3 while decompressing data: invalid block type

INFO Processing status: 40 for Task [4/27/19 - Combined] (876e9ede-7233-4516-bd4f-53ccbf6b8b3f)
[2019-04-30 10:07:28,393: INFO/ForkPoolWorker-1129] Processing status: 40 for Task [4/27/19 - Combined] (876e9ede-7233-4516-bd4f-53ccbf6b8b3f)
INFO Removing old assets directory: /webodm/app/media/project/1/task/876e9ede-7233-4516-bd4f-53ccbf6b8b3f/assets/ for Task [4/27/19 - Combined] (876e9ede-7233-4516-bd4f-53ccbf6b8b3f)
[2019-04-30 10:07:28,393: INFO/ForkPoolWorker-1129] Removing old assets directory: /webodm/app/media/project/1/task/876e9ede-7233-4516-bd4f-53ccbf6b8b3f/assets/ for Task [4/27/19 - Combined] (876e9ede-7233-4516-bd4f-53ccbf6b8b3f)
INFO Downloading all.zip for Task [4/27/19 - Combined] (876e9ede-7233-4516-bd4f-53ccbf6b8b3f)
[2019-04-30 10:07:28,692: INFO/ForkPoolWorker-1129] Downloading all.zip for Task [4/27/19 - Combined] (876e9ede-7233-4516-bd4f-53ccbf6b8b3f
`

pierotofy, 2019-05-01:
Hey @djfurie thanks for reporting this!

As a workaround, you should be able to import the `all.zip` file into WebODM by pressing the arrow next to "Select Images and GCPs" and upload it directly from your computer or by specifying a URL.

pierotofy, 2019-06-14:
This should now be mitigated by recent changes that retry the download multiple times. If this happens again, please re-open the issue!

closed 2019-06-14 by pierotofy