WebODM/WebODM #660: Plugin Idea: allow users to load already processed orthophotos, DEMs and 3D models

pierotofy, 2019-05-02:
Seems like a recurring use case for some users.

kikislater, 2019-05-06:
This function is already present in MicMac for information for DEM. It allows to process, clean dem or point cloud, import it to have a clean Ortho !

smathermather, 2019-05-06:
@kikislater -- have you seen this?: https://github.com/dronemapper-io/NodeMICMAC

kikislater, 2019-05-07:
Yes, I have to find time to test it nos ;-)

kikislater, 2019-05-07:
As we discuss about it, will it be plan in the near future to choose between opensfm and micmac ?

pierotofy, 2019-05-07:
We'll probably add the option to add a NodeMicMac node when running WebODM, yes.

smathermather, 2019-05-07:
It makes lots of sense.

pierotofy, 2020-04-04:
This will likely be included as part of core, as the dependency to PotreeConverter/Entwine is difficult to add as a plugin (for point clouds).

byhkalo, 2021-04-20:
Hello @pierotofy, @smathermather

I see this issue and this problem is actual for me. I have a lot of orthophotos from Pix4D (I processed up to 1000 images). I would like to import them to present to clients.

How can I support that feature? I don't have experience in the web and can't write code by myself. But can I pay or something like that to support your team. How much you need?

byhkalo, 2021-04-22:
Hello @pierotofy and @smathermather!
Hope you are going well.
What do you think about that feature? Is it hard to implement?

pierotofy, 2021-04-22:
Hey @byhkalo :hand: it's not difficult, but just time consuming (making the UI, API endpoints, async processing of point clouds to generate EPTs). Get in touch at https://uav4geo.com/contact we can discuss time/cost estimates.