WebODM/WebODM #661: Task fails during compression the first time then succeeds

ZInventor, 2019-05-07:
### How did you install WebODM? (Docker, natively, ...)?

Natively on dedicated clean install of Ubuntu

### What's your browser and operating system?

Browser is chrome on Win10 enterprise

### What is the problem?

Processing fails at the final step during compression (after generating all points) on large datasets. Plenty of RAM left (32 gb in the machine, 14 in use during compression) and plenty of HDD left (2TB SSD as main drive). If I re-run from orthophoto after the glitch, it completes as expected with no issues.

### What should be the expected behavior?

Complete processing on the first try

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

WEBODM parameters: "pc-classify: true, orthophoto-resolution: 2.0, dtm: true, dem-resolution: 2.0, dsm: true, depthmap-resolution: 1000"

This seems to happen more often on very large datasets (500+ images), and when multiple datasets are being processed simultaneously (1000+ total images).

ZInventor, 2019-05-08:
The status box turned red and showed an error, I will try and replicate the issue again and get the log file.

dronesperhour, 2019-05-14:
I believe I experienced the same error reading "invalid zip file". Running WebODM with Docker on macOS High Sierra, with options "orthophoto-resolution: 50, matcher-distance: 50". I've also been running 2 tasks simultaneously (98 / 246 pictures). Using "rerun-from: odm_orthophoto" completed successfully without changing anything.

PR #663 (2019-05-23): Sprint
 - Retry zip file downloads at task completion (should fix #661 )
closed 2019-05-23 by pierotofy