WebODM/WebODM #662: [Improvement] Make progress reporting / restart stages based on common interface

pierotofy, 2019-05-09:
Currently WebODM looks at the strings in the console output to determine the current stage and progressbar values. This is obviously not going to work for engines other than ODM (e.g. MicMac).

Stuff like: https://github.com/OpenDroneMap/WebODM/blob/master/app/static/app/js/classes/PipelineSteps.js#L7

And https://github.com/OpenDroneMap/WebODM/blob/master/app/models/task.py#L179

Needs to follow values coming from the NodeODM API. https://github.com/OpenDroneMap/NodeODM/issues/80

@stoermerjp if the progress bars don't work when you plug NodeMicMac, it's expected at this point! Also the "Basic View" will probably be a bit messed up in the UI when you create a task. Processing should work however.

Somewhat related, the various "Restart" options are also ODM-oriented. I assume that MicMac can also resume processing from various stages? Sometimes those are useful if a user tweaks a variable that influences only a certain stage (especially later stages), so that they don't have to re-run the entire pipeline.

pierotofy, 2019-05-09:
This should probably supersede #655

ChipwizBen, 2019-05-09:
Is it also safe to assume that this would also fix the issue of WebODM losing track of a running task in Node if WebODM is restarted? I was about to raise an Issue about it but I won't if it essentially duplicates this issue.

pierotofy, 2019-05-09:
I think so.

dronemapper-io, 2019-05-09:
@pierotofy Agreed. Yes, MM can restart at certain stages too. Each step (camera calibration/dem construction/ortho) will output a lot of strings to stdout. We could certainly parse this output to provide some indication of progress. I will investigate the two links you sent and give it some thought -- definitely doable and a nice improvement

pierotofy, 2019-05-23:
Progress reporting is now done! Stages restart will be a separate issue.

closed 2019-05-23 by pierotofy