WebODM/WebODM #674: Possible SW Bug - Loading Photo Files

mgaviria-m, 2019-06-12:
Docker install via the start script (/webodm.sh start --with-micmac)

Browser and operating system? 

Safari 12.1.1.  &  Firefox 66.0.5. on OSX 10.14.5 (both equal)

The problem ==> An error box (red) appears when trying to load the Photos for a project. 
The text warns about file types differentiations from jpg. and the process is aborted.
after verification all of them are valid jpg.

The expected behavior ==> Successful load of files

 We can reproduce this ==>  Trying ring to loading files into a new project .

Thanks

pierotofy, 2019-06-12:
Hi @mgaviria-m,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-06-15 13:00) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2019-06-14:
Hey @mgaviria-m :hand: thanks for reporting. Do you have a copy of the images uploaded somewhere so that we could take a look?

pierotofy, 2020-04-02:
Closing due to inability to reproduce.

closed 2020-04-02 by pierotofy