WebODM/WebODM #675: Image groups are not supported in WebODM

pierotofy, 2019-06-13:
There's currently no way to use image groups in split merge via WebODM.

pierotofy, 2019-06-13:
Hi @pierotofy,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-06-16 13:25) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

smathermather, 2019-06-14:
Hmm, good point.

astrogazer63, 2020-09-30:
Has this changed or been updated with more recent versions???

smathermather, 2020-09-30:
It has not yet been addressed in more recent versions.

pierotofy, 2021-02-05:
This is being addressed in https://github.com/OpenDroneMap/NodeODM/pull/143 :clinking_glasses:

pierotofy, 2021-04-19:
Done!

closed 2021-04-19 by pierotofy