WebODM/WebODM #678: ImportError: No module named opensfm.exif

ChipwizBen, 2019-06-17:
### How did you install WebODM? (Docker, natively, ...)?

Native

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

N/A

### What is the problem?

Seconds after starting the job dies due to a missing opensfm module:
```
File "/code/run.py", line 12, in <module>
from stages.odm_app import ODMApp
File "/code/stages/odm_app.py", line 4, in <module>
from opendm import types
File "/code/opendm/types.py", line 6, in <module>
from opensfm.exif import sensor_string
```

I've tried to update twice with the 1.3.4 script, and twice with the newer 1.4.0 script, but the issue is not corrected.

### What should be the expected behavior?

Job should not die.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Run the update on a native install.

ChipwizBen, 2019-06-17:
From syslog:

```
Jun 17 14:02:48 proc02.nwk1.com node[1774]: info: About to run: /code/run.sh --rerun-from dataset --fast-orthophoto --project-path /www/data 3dae2856-72c1-4f58-b07c-86cd326589a9
Jun 17 14:02:49 proc02.nwk1.com node[1774]: warn:  message=ENOENT: no such file or directory, open 'data/3dae2856-72c1-4f58-b07c-86cd326589a9/images.json', stack=Error: ENOENT: no such file or directory, open 'data/3dae2856-72c1-4f58-b07c-86cd326589a9/images.json', errno=-2, code=ENOENT, syscall=open, path=data/3dae2856-72c1-4f58-b07c-86cd326589a9/images.json
```

ChipwizBen, 2019-06-17:
Nodes on 0.6.0 still work correctly. The issue started from 0.7.0.

pierotofy, 2019-06-17:
OpenSfM must have failed to build properly (or did not download at all).

What does this say?

```
cd /code/Superbuild/build
make opensfm
```

?

ChipwizBen, 2019-06-17:
```
root@proc02:/code/SuperBuild/build# make
[  6%] Built target dem2points
[ 13%] Built target mve
[ 19%] Built target mvstexturing
[ 25%] Built target laszip
[ 31%] Built target hexer
[ 38%] Built target zstd
[ 44%] Built target pdal
[ 45%] Performing configure step for 'entwine'
-- Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the system variable OPENSSL_ROOT_DIR (missing:  OPENSSL_LIBRARIES OPENSSL_INCLUDE_DIR) (Required is at least version "1.0.1")
OpenSSL NOT found - `export OPENSSL_ROOT_DIR=___`
Google storage IO will not be available
CMake Warning (dev) at cmake/pdal.cmake:3 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target "pdalcpp".
Call Stack (most recent call first):
  CMakeLists.txt:37 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

Configuring with NO unit tests
--
-- The following OPTIONAL packages have been found:

 * Backtrace
 * Curl
 * PkgConfig

-- The following REQUIRED packages have been found:

 * PDAL (required version >= 1.9.1)
 * Threads

-- The following OPTIONAL packages have not been found:

 * OpenSSL (required version >= 1.0.1)

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
/code/SuperBuild/src/entwine/app/OPENSSL_INCLUDE_DIR
   used as include directory in directory /code/SuperBuild/src/entwine/app
/code/SuperBuild/src/entwine/entwine/builder/OPENSSL_INCLUDE_DIR
   used as include directory in directory /code/SuperBuild/src/entwine/entwine/builder
/code/SuperBuild/src/entwine/entwine/formats/cesium/OPENSSL_INCLUDE_DIR
   used as include directory in directory /code/SuperBuild/src/entwine/entwine/formats/cesium
/code/SuperBuild/src/entwine/entwine/io/OPENSSL_INCLUDE_DIR
   used as include directory in directory /code/SuperBuild/src/entwine/entwine/io
/code/SuperBuild/src/entwine/entwine/reader/OPENSSL_INCLUDE_DIR
   used as include directory in directory /code/SuperBuild/src/entwine/entwine/reader
/code/SuperBuild/src/entwine/entwine/third/OPENSSL_INCLUDE_DIR
   used as include directory in directory /code/SuperBuild/src/entwine/entwine/third
/code/SuperBuild/src/entwine/entwine/types/OPENSSL_INCLUDE_DIR
   used as include directory in directory /code/SuperBuild/src/entwine/entwine/types
/code/SuperBuild/src/entwine/entwine/util/OPENSSL_INCLUDE_DIR
   used as include directory in directory /code/SuperBuild/src/entwine/entwine/utilOPENSSL_CRYPTO_LIBRARY (ADVANCED)
    linked by target "entwine" in directory /code/SuperBuild/src/entwine
OPENSSL_SSL_LIBRARY (ADVANCED)
    linked by target "entwine" in directory /code/SuperBuild/src/entwine

-- Configuring incomplete, errors occurred!
See also "/code/SuperBuild/build/entwine/CMakeFiles/CMakeOutput.log".
See also "/code/SuperBuild/build/entwine/CMakeFiles/CMakeError.log".
CMakeFiles/entwine.dir/build.make:109: recipe for target 'entwine/stamp/entwine-configure' failed
make[2]: *** [entwine/stamp/entwine-configure] Error 1
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/entwine.dir/all' failed
make[1]: *** [CMakeFiles/entwine.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
[... 43 lines cut ...]
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /code/SuperBuild/build/entwine/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_7d0ae/fast"
make[3]: Entering directory '/code/SuperBuild/build/entwine/CMakeFiles/CMakeTmp'
/usr/bin/make -f CMakeFiles/cmTC_7d0ae.dir/build.make CMakeFiles/cmTC_7d0ae.dir/build
make[4]: Entering directory '/code/SuperBuild/build/entwine/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_7d0ae.dir/CheckFunctionExists.c.o
/usr/bin/cc    -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_7d0ae.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.5/Modules/CheckFunctionExists.c
Linking C executable cmTC_7d0ae
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_7d0ae.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTC_7d0ae.dir/CheckFunctionExists.c.o  -o cmTC_7d0ae -rdynamic -lpthreads
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_7d0ae.dir/build.make:97: recipe for target 'cmTC_7d0ae' failed
make[4]: *** [cmTC_7d0ae] Error 1
make[4]: Leaving directory '/code/SuperBuild/build/entwine/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_7d0ae/fast' failed
make[3]: *** [cmTC_7d0ae/fast] Error 2
make[3]: Leaving directory '/code/SuperBuild/build/entwine/CMakeFiles/CMakeTmp'

root@proc02:/code/SuperBuild/build#
```

ChipwizBen, 2019-06-17:
OpenSSL:
```
root@proc02:/code/SuperBuild/build# dpkg -l | grep openssl
ii  libgnutls-openssl27:amd64              3.4.10-4ubuntu1.5                          amd64        GNU TLS library - OpenSSL wrapper
ii  libxmlsec1-openssl                     1.2.20-2ubuntu4                            amd64        Openssl engine for the XML security library
ii  openssl                                1.0.2g-1ubuntu4.15                         amd64        Secure Sockets Layer toolkit - cryptographic utility
ii  python-openssl                         0.15.1-2ubuntu0.2                          all          Python 2 wrapper around the OpenSSL library
root@proc02:/code/SuperBuild/build#
```

ChipwizBen, 2019-06-17:
Doh, output for `make opensdm` too:

```
[ 56%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/mathop_sse2.c.o
[ 58%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/mser.c.o
[ 60%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/pgm.c.o
[ 62%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/quickshift.c.o
[ 64%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/random.c.o
[ 66%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/rodrigues.c.o
[ 68%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/scalespace.c.o
[ 70%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/sift.c.o
[ 72%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/slic.c.o
[ 75%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/stringop.c.o
[ 77%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/svm.c.o
[ 79%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/svmdataset.c.o
[ 81%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/vlad.c.o
[ 83%] Linking C static library libvl.a
[ 83%] Built target vl
Scanning dependencies of target bundle
[ 85%] Building CXX object CMakeFiles/bundle.dir/bundle/src/bundle_adjuster.cc.o
[ 87%] Linking CXX static library libbundle.a
[ 87%] Built target bundle
Scanning dependencies of target csfm
[ 89%] Building CXX object CMakeFiles/csfm.dir/csfm.cc.o
[ 91%] Building CXX object CMakeFiles/csfm.dir/types.cc.o
[ 93%] Building CXX object CMakeFiles/csfm.dir/akaze_bind.cc.o
[ 95%] Building CXX object CMakeFiles/csfm.dir/matching.cc.o
[ 97%] Building CXX object CMakeFiles/csfm.dir/depthmap.cc.o
[100%] Linking CXX shared module /code/SuperBuild/src/opensfm/opensfm/csfm.so
[100%] Built target csfm
[ 96%] No install step for 'opensfm'
[100%] Completed 'opensfm'
[100%] Built target opensfm
root@proc02:/code/SuperBuild/build# echo $?
0
root@proc02:/code/SuperBuild/build#
```

pierotofy, 2019-06-17:
Mm, strange. 

Are you installing other software on this server alongside WebODM?

ChipwizBen, 2019-06-17:
Not really, just basic server management stuff from default repos:
```
apt-get -y install snmpd rsyslog ntp dstat
```

And Ceph for the mounts:
```
wget -q -O- 'https://download.ceph.com/keys/release.asc' | sudo apt-key add -
apt-add-repository "deb https://download.ceph.com/debian-luminous/ $(lsb_release -sc) main"
apt-get clean all
apt-get update
apt-get -y install ceph-common
```

Have been running with this config for months.

pierotofy, 2019-06-18:
Seems like a problem related to cmake, but I don't know what caused it. https://stackoverflow.com/questions/31948521/building-error-using-cmake-cannot-find-lpthreads

ChipwizBen, 2019-06-20:
I was able to replicate this on a test node without Ceph. It's fixed by running your suggestion but must be run manually at every update:
```
cd /code/SuperBuild/build
make opensfm
```

pierotofy, 2020-04-02:
Closing this since it's related to ODM install, not WebODM. If this is still a problem I'd recommend opening an updated issue on the ODM repo. :+1:

closed 2020-04-02 by pierotofy