WebODM/WebODM #679: Native install locked at Potree 1.5RC

ChipwizBen, 2019-06-17:
### How did you install WebODM? (Docker, natively, ...)?

Native

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

N/A

### What is the problem?

Potree is stuck at version 1.5RC, while the Docker version has moved on to 1.6. There may be other things that are not kept up to date, but Potree is an obvious one due to the noted version number in the left pane.

I've tried to update twice with the 1.3.4 script, and twice with the newer 1.4.0 script, but the issue is not corrected.

### What should be the expected behavior?

Native and Docker installs should track together.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Update native install.

ChipwizBen, 2019-06-17:
Traced this to /webodm:

```
Updating 0d485fc..abc555c
error: Your local changes to the following files would be overwritten by merge:
	package.json
Please, commit your changes or stash them before you can merge.
```

I haven't modified that file so it must have been WebODM. The updater doesn't halt on $? != 0 errors (it reports a successful update at the end). I fixed it with a `git reset --hard`, but I guess the better fix would be to update the script to halt on errors otherwise they become silent in the noise of the 10,000+ lines of output from the update script.

pierotofy, 2019-06-17:
Thanks @ChipwizBen for the report.

pierotofy, 2020-04-02:
Took note of this a while ago, so this can be closed.

closed 2020-04-02 by pierotofy