WebODM/WebODM #681: 3-7 tasks submitted for each job

ChipwizBen, 2019-06-19:
### How did you install WebODM? (Docker, natively, ...)?

Native

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox

### What is the problem?

When either submitting a new task, or restarting an old or failed task, up to 7 of these tasks are sent to ODM. These are _not_ reflected in the UI other than under Processing Nodes where the task count is obviously higher than expected. The issue started to occur after the update to 0.7.0. These tasks are viewable in tasks.json on each of the nodes.

syslog:
```
Jun 19 21:42:38 control.nwk1.com celery[1748]: INFO Automatically assigned processing node proc06.nwk1.com to Task [CPMS Map (Buildings)] (ce06b61c-0a25-475a-9a04-6fa4d88622cd)
Jun 19 21:42:38 control.nwk1.com celery[1748]: [2019-06-19 21:42:38,462: INFO/ForkPoolWorker-6] Automatically assigned processing node proc06.nwk1.com to Task [CPMS Map (Buildings)] (ce06b61c-0a25-475a-9a04-6fa4d88622cd)
Jun 19 21:42:38 control.nwk1.com celery[1748]: INFO Processing... Task [CPMS Map (Buildings)] (ce06b61c-0a25-475a-9a04-6fa4d88622cd)
Jun 19 21:42:38 control.nwk1.com celery[1748]: [2019-06-19 21:42:38,474: INFO/ForkPoolWorker-6] Processing... Task [CPMS Map (Buildings)] (ce06b61c-0a25-475a-9a04-6fa4d88622cd)
Jun 19 21:42:46 control.nwk1.com celery[1748]: INFO Processing... Task [CPMS Map (Buildings)] (ce06b61c-0a25-475a-9a04-6fa4d88622cd)
Jun 19 21:42:46 control.nwk1.com celery[1748]: [2019-06-19 21:42:46,479: INFO/ForkPoolWorker-8] Processing... Task [CPMS Map (Buildings)] (ce06b61c-0a25-475a-9a04-6fa4d88622cd)
Jun 19 21:42:51 control.nwk1.com celery[1748]: INFO Restarting Task [VF Earthworks 1 (Cloned)] (1932b742-c5b6-4099-99f1-f99bf79e8ee3)
Jun 19 21:42:51 control.nwk1.com celery[1748]: [2019-06-19 21:42:51,603: INFO/ForkPoolWorker-7] Restarting Task [VF Earthworks 1 (Cloned)] (1932b742-c5b6-4099-99f1-f99bf79e8ee3)
Jun 19 21:42:51 control.nwk1.com celery[1748]: INFO Task [VF Earthworks 1 (Cloned)] (1932b742-c5b6-4099-99f1-f99bf79e8ee3) needs to be reprocessed
Jun 19 21:42:51 control.nwk1.com celery[1748]: [2019-06-19 21:42:51,616: INFO/ForkPoolWorker-7] Task [VF Earthworks 1 (Cloned)] (1932b742-c5b6-4099-99f1-f99bf79e8ee3) needs to be reprocessed
Jun 19 21:42:56 control.nwk1.com celery[1748]: INFO Processing... Task [VF Earthworks 1 (Cloned)] (1932b742-c5b6-4099-99f1-f99bf79e8ee3)
Jun 19 21:42:56 control.nwk1.com celery[1748]: [2019-06-19 21:42:56,532: INFO/ForkPoolWorker-9] Processing... Task [VF Earthworks 1 (Cloned)] (1932b742-c5b6-4099-99f1-f99bf79e8ee3)
Jun 19 21:43:05 control.nwk1.com snmpd[1738]: error on subcontainer 'ia_addr' insert (-1)
Jun 19 21:43:11 control.nwk1.com celery[1748]: INFO Processing... Task [VF Earthworks 1 (Cloned)] (1932b742-c5b6-4099-99f1-f99bf79e8ee3)
Jun 19 21:43:11 control.nwk1.com celery[1748]: [2019-06-19 21:43:11,489: INFO/ForkPoolWorker-7] Processing... Task [VF Earthworks 1 (Cloned)] (1932b742-c5b6-4099-99f1-f99bf79e8ee3)
Jun 19 21:43:16 control.nwk1.com celery[1748]: INFO Processing... Task [CPMS Map (Buildings)] (ce06b61c-0a25-475a-9a04-6fa4d88622cd)
Jun 19 21:43:16 control.nwk1.com celery[1748]: [2019-06-19 21:43:16,434: INFO/ForkPoolWorker-12] Processing... Task [CPMS Map (Buildings)] (ce06b61c-0a25-475a-9a04-6fa4d88622cd)
Jun 19 21:43:26 control.nwk1.com celery[1748]: INFO Processing... Task [VF Earthworks 1 (Cloned)] (1932b742-c5b6-4099-99f1-f99bf79e8ee3)
Jun 19 21:43:26 control.nwk1.com celery[1748]: [2019-06-19 21:43:26,484: INFO/ForkPoolWorker-11] Processing... Task [VF Earthworks 1 (Cloned)] (1932b742-c5b6-4099-99f1-f99bf79e8ee3)
Jun 19 21:43:35 control.nwk1.com snmpd[1738]: error on subcontainer 'ia_addr' insert (-1)
Jun 19 21:43:36 control.nwk1.com celery[1748]: INFO Processing... Task [CPMS Map (Buildings)] (ce06b61c-0a25-475a-9a04-6fa4d88622cd)
Jun 19 21:43:36 control.nwk1.com celery[1748]: [2019-06-19 21:43:36,423: INFO/ForkPoolWorker-13] Processing... Task [CPMS Map (Buildings)] (ce06b61c-0a25-475a-9a04-6fa4d88622cd)
Jun 19 21:43:41 control.nwk1.com celery[1748]: INFO Processing... Task [VF Earthworks 1 (Cloned)] (1932b742-c5b6-4099-99f1-f99bf79e8ee3)
Jun 19 21:43:41 control.nwk1.com celery[1748]: [2019-06-19 21:43:41,469: INFO/ForkPoolWorker-10] Processing... Task [VF Earthworks 1 (Cloned)] (1932b742-c5b6-4099-99f1-f99bf79e8ee3)
Jun 19 21:44:01 control.nwk1.com CRON[3026]: (root) CMD (   test -x /etc/cron.daily/popularity-contest && /etc/cron.daily/popularity-contest --crond)
Jun 19 21:44:02 control.nwk1.com celery[1748]: INFO Processing... Task [CPMS Map (Buildings)] (ce06b61c-0a25-475a-9a04-6fa4d88622cd)
Jun 19 21:44:02 control.nwk1.com celery[1748]: [2019-06-19 21:44:02,942: INFO/ForkPoolWorker-9] Processing... Task [CPMS Map (Buildings)] (ce06b61c-0a25-475a-9a04-6fa4d88622cd)
Jun 19 21:44:05 control.nwk1.com snmpd[1738]: error on subcontainer 'ia_addr' insert (-1)
Jun 19 21:44:35 control.nwk1.com snmpd[1738]: error on subcontainer 'ia_addr' insert (-1)
Jun 19 21:44:54 control.nwk1.com celery[1748]: INFO Processing... Task [CPMS Map (Buildings)] (ce06b61c-0a25-475a-9a04-6fa4d88622cd)
Jun 19 21:44:54 control.nwk1.com celery[1748]: [2019-06-19 21:44:54,864: INFO/ForkPoolWorker-7] Processing... Task [CPMS Map (Buildings)] (ce06b61c-0a25-475a-9a04-6fa4d88622cd)
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]: ERROR Internal Server Error: /api/projects/424/tasks/1932b742-c5b6-4099-99f1-f99bf79e8ee3/
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]: Traceback (most recent call last):
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]:   File "/webodm/python3-venv/lib/python3.5/site-packages/django/contrib/sessions/backends/base.py", line 190, in _get_session
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]:     return self._session_cache
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]: AttributeError: 'SessionStore' object has no attribute '_session_cache'
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]: During handling of the above exception, another exception occurred:
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]: Traceback (most recent call last):
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]:   File "/webodm/python3-venv/lib/python3.5/site-packages/django/db/backends/base/base.py", line 216, in ensure_connection
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]:     self.connect()
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]:   File "/webodm/python3-venv/lib/python3.5/site-packages/django/db/backends/base/base.py", line 194, in connect
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]:     self.connection = self.get_new_connection(conn_params)
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]:   File "/webodm/python3-venv/lib/python3.5/site-packages/django/db/backends/postgresql/base.py", line 178, in get_new_connection
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]:     connection = Database.connect(**conn_params)
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]:   File "/webodm/python3-venv/lib/python3.5/site-packages/psycopg2/__init__.py", line 130, in connect
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]:     conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]: psycopg2.OperationalError: FATAL:  sorry, too many clients already
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]: FATAL:  sorry, too many clients already
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]: The above exception was the direct cause of the following exception:
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]: Traceback (most recent call last):
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]:   File "/webodm/python3-venv/lib/python3.5/site-packages/django/core/handlers/exception.py", line 34, in inner
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]:     response = get_response(request)
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]:   File "/webodm/python3-venv/lib/python3.5/site-packages/django/utils/deprecation.py", line 90, in __call__
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]:     response = self.process_request(request)
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]:   File "/webodm/python3-venv/lib/python3.5/site-packages/django/middleware/locale.py", line 21, in process_request
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]:     language = translation.get_language_from_request(request, check_path=i18n_patterns_used)
[... 27 lines cut ...]
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]:     self.ensure_connection()
[... cut ...]
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]: django.db.utils.OperationalError: FATAL:  sorry, too many clients already
Jun 19 21:44:59 control.nwk1.com gunicorn[1743]: FATAL:  sorry, too many clients already

```

### What should be the expected behavior?

One task submitted should equal one task processed.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Update to 0.7.0. Submit or restart a task.

pierotofy, 2019-06-19:
Confirm this is a problem. Investigating possible causes. Thanks for reporting it.

pierotofy, 2019-06-19:
Please update, the problem should be fixed. If it isn't feel free to re-open.

PR #684 (2019-06-19): BUG: lock deletion fix
Pretty serious bug fix, fixes #681
closed 2019-06-19 by pierotofy