WebODM/WebODM #682: Hang between Upload and Processing

saffa19, 2019-06-19:
#### How did you install WebODM?

Docker

#### What's your browser and operating system?

Chrome 74 on Windows 10

#### What is the problem?

Processing hangs after resizing and upload. No real activity (CPU usage <5%) in any of the Docker containers (seen with `docker stats`).

#### What should be the expected behavior?

Processing continues to next step (matching etc).

#### How can we reproduce this?

Unfortunately can't include dataset (255 .JPG files, 5472 x 3648).
No GCP file.
"Fast Orthophoto" processing option enabled with Resize to 2048px.
Will go through the process again to add console log up to the point of hanging.

#### My workaround

Resize in Browser to 2048px
"Default" processing option.

pierotofy, 2019-06-19:
Hey @saffa19 :hand: could you confirm that you are running version 1.0.2 from the diagnostic menu?

Version 1.0.1 has just been fixed (see #681) which could be causing all sorts of troubles.

saffa19, 2019-06-20:
Hi Piero, yes I was running 1.0.1 - will update and test again!

saffa19, 2019-06-20:
Fixed with 1.0.2

closed 2019-06-20 by saffa19