WebODM/WebODM #683: Update documentation for default nodeodm processing node

saffa19, 2019-06-19:
Readme hasn't been updated since the default processing node was removed from docker-compose file.
The comment in docker-compose.nodeodm.yml says that the default node makes it easier for new users so it would be useful to tell people how to use it in the readme.

Just a line saying that you can use `$ docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up` to use the host machine as the processing node

pierotofy, 2019-06-19:
Hi @saffa19,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-06-22 09:55) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2019-06-19:
Hey @saffa19 you typically do not need to use docker-compose directly, that's what the `webodm.sh` script is for. Are you perhaps on Windows?

saffa19, 2019-06-20:
Yes, I've been running it from powershell

pierotofy, 2019-06-20:
Yeah, I wouldn't recommend using powershell. Use "Git Bash" instead and use the `./webodm.sh` script.

Please feel free to update the documentation in both the README and on https://github.com/OpenDroneMap/docs if you think it could help people in the future!

I will close this for now, please open a PR with proposed documentation changes if you want.

closed 2019-06-20 by pierotofy