WebODM/WebODM #686: Download to File console log sometimes does not work

pierotofy, 2019-06-20:
Seems to affect some Chrome users in certain cases. No reproducible test case available as of yet.

pierotofy, 2019-06-20:
Hi @pierotofy,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-06-22 21:55) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

danbjoseph, 2019-06-21:
[image]
[image]
[image]

is there other info i can provide?

pierotofy, 2019-06-21:
Yes please, length of the task log (in number of lines, you can see the full task log from the "Administration" -- "Application" menu by navigating under "Tasks") and operating system could help reproduce this.

Thanks!!

danbjoseph, 2019-06-21:
[image]
(WebODM is running on an Intel NUC via POSM)

-----------

[Lipata_1.txt](https://github.com/OpenDroneMap/WebODM/files/3312721/Lipata_1.txt) is the text log saved using Firefox

pierotofy, 2019-06-25:
This is fixed in #688. Thanks for reporting!

closed 2019-06-25 by pierotofy

danbjoseph, 2019-06-25:
Thanks for fixing it!