WebODM/WebODM #692: Support docker swarm deployments

bubthegreat, 2019-07-03:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For questions, comments and feature requests, please open a topic on http://community.opendronemap.org/c/webodm. 

Also, is your problem listed here? https://github.com/OpenDroneMap/WebODM#common-troubleshooting
If so, do not open an issue. We already know about it and steps to solve it are already documented.

Every issue needs to follow the format specified below. Do not remove the questions, simply answer each question by typing below it. If you don't know how to answer a question, write so. Questions that don't follow these guidelines will be automatically closed by our friendly IssueBot (who's not very smart and needs to see a particular format to keep an issue open).
****************************************

### How did you install WebODM? (Docker, natively, ...)?

Git clone, and docker-compose up -d - that works fine

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Not a browser or platform issue

### What is the problem?

Would like ODM to be deployable to a swarm stack

### What should be the expected behavior?

single command to deploy a scalable stack via docker stack deploy, i.e.:
`docker stack deploy -c docker-compose.yml webodm`

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Try to deploy via docker stack deploy and it'll fail, because it's not designed for a stack deployment.  Stack deployment would preferably be in a later compose version (3.x+ rather than early 2.x - there are a lot of improvements since then)

pierotofy, 2019-07-04:
Hey @bubthegreat, that would be a cool addition! Would you have an interest in contributing to it?

bubthegreat, 2019-07-04:
Absolutely!  I'll pull the repo and see what I need to do to get it to
work.  Will post questions here unless theres a better place for it.

pierotofy, 2019-07-04:
That's fantastic! :+1:  This thread is fine for questions.

bubthegreat, 2019-07-05:
Allright - initial questions:

1. Is there an entrypoint for dynamic addition of nodeodm nodes, or is that currently done manually?  It looks like we're statically defining the hostname for the initial (node-odm-1) in the start script - this might make it challenging to scale workers if they don't have an automated way of joining a cluster.  Is there a join key that we create somewhere that we can define and share across all of them so we can dynamically add a new node if we scale them, or is this just done manually right now?
2. Any reasons we're defining the entrypoint in the compose file that I should be aware of, or is that something we can move to the image and let folks override the "start"

pierotofy, 2019-07-10:
1. It's currently done via docker-compose (1 node max, no possibility to add more).
2. Assuming you are referring to the webodm_webapp image, we use an entrypoint in docker-compose because we spawn two containers, one is the actual webapp, the other is a background worker. There's probably other ways to set that up (have two separate images with different entrypoints?) I'm guessing.

bubthegreat, 2019-07-15:
Looks like I'm going to have a baby instead and probably won't be able to
get to this for a while, but if nobody else picks it up I'll jump back on
it as my time frees up.

pierotofy, 2019-07-15:
Congrats! I can understand the shift in priorities :smile: I'll leave this open for future development.

hefnawi, 2019-10-19:
Hello,

I will start tackling this issue and keep you updated guys.

Thanks,
Ahmed Hefnawi

hefnawi, 2019-11-03:
Still working my way slowly through webODM and in parallel working on the docker stack yaml file, however, I will just ask questions here if you don't mind whenever I get stuck =)

pierotofy, 2019-11-04:
Of course @hefnawi , thanks for the update! :+1: Let us know if you have questions.

Ghost7K, 2020-05-25:
@hefnawi i also tried to deploy it to a stack, and now that i found this thread i was wondering if you had success with Docker 3x Swarm deploay. Greetings

bubthegreat, 2020-05-27:
Current setup does not lend it self to swarm deployment, no - the dockerfiles would need to be rewritten for swarm/kubernetes configs.

antonyharfield, 2020-06-07:
I'm interested in deploying to Kubernetes and can help out translating files if needed -- has any work been started on this already?

dimm0, 2020-11-12:
Working ok for us in kubernetes: https://gitlab.nautilus.optiputer.net/prp/webodm
I found that there's a horizontally scalable version now. But still no support for kubernetes

pierotofy, 2023-01-28:
I'm going to close this; if anyone wants to contribute a PR for this, it would be most welcome. :pray:

closed 2023-01-28 by pierotofy