WebODM/WebODM #699: HTTP status codes could be better? (for orthophoto TMS layer)

danbjoseph, 2019-07-15:
### How did you install WebODM? (Docker, natively, ...)?

as outlined in [Become a Contributor](https://github.com/OpenDroneMap/WebODM#become-a-contributor), starting it with `./webodm.sh start --dev`

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

macOS High Sierra 10.13.6 (17G7024)
Docker Desktop Community 2.0.0.3 (31259)

### What is the problem?

A valid URL for a tile returns a 404 when not logged in.

### What should be the expected behavior?

I would expect a 403 error? I was having trouble figuring out why I couldn't load imagery into JOSM while working on #697.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Get a URL for a tile. Then try and get it from a different browser where you are not logged into WebODM or from another application. In the example screengrabs below,  http://localhost:8000/api/projects/1/tasks/800373e4-c23a-4c3c-af41-85da6f769168/orthophoto/tiles/20/256336/679092.png worked in Chrome (where I was logged in to WebODM), but not in Firefox or with `wget`.

[image]

[image]

danbjoseph, 2019-07-15:
Cool. It does work if you enable sharing of the task.
[image]

Is there a better way to inform the user other than a `window.confirm()` on clicking to edit in JOSM? What's a good notification message?

[image]

pierotofy, 2019-07-15:
Actually, this is a security feature by design. If we return a 403 error, a malicious user could use error codes to discover UUIDs (non existing project = 404, unauthorized = 403). It's a fairly common practice in security.

danbjoseph, 2019-07-15:
ah, interesting. makes sense. 👍

closed 2019-07-15 by danbjoseph