WebODM/WebODM #703: Cannot start WebODM-db exited with code 1

silverefendy, 2019-07-17:
### What is the problem?

I cannot start WebODM again.

### What should be the expected behavior?

It should start as usually as good as always.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Last time i process about 800 photos. And got some error. 
My ubuntu slow to freeze and then i hard reset it using small button near power.
After trying to restart, got this error.
And after "./webodm.sh update" command, still got this error.

drone@drone-comp:~/WebODM$ sudo ./webodm.sh restart
Checking for docker...   OK
Checking for git...   OK
Checking for python...   OK
Checking for pip...   OK
Checking for docker-compose...   OK
Restarting WebODM...
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml -f docker-compose.nodemicmac.yml down --remove-orphans
Removing webapp     ... done
Removing worker     ... done
Removing node-odm-1 ... done
Removing db         ... done
Removing broker     ... done
Removing network webodm_default
Starting WebODM...

Using the following environment:
================================
Host: localhost
Port: 8000
Media directory: appmedia
SSL: NO
SSL key: 
SSL certificate: 
SSL insecure port redirect: 80
Celery Broker: redis://broker
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up
Starting node-odm-1 ... done
Starting broker     ... done
Starting db         ... done
Starting worker     ... done
Starting webapp     ... done
ERROR: No containers to start
Creating network "webodm_default" with the default driver
Creating db         ... done
Creating node-odm-1 ... done
Creating broker     ... done
Creating worker     ... done
Creating webapp     ... done
Attaching to db, broker, node-odm-1, worker, webapp
db            | LOG:  database system was shut down at 2019-07-16 09:42:31 UTC
db            | LOG:  invalid primary checkpoint record
db            | LOG:  invalid secondary checkpoint record
db            | PANIC:  could not locate a valid checkpoint record
db            | LOG:  startup process (PID 23) was terminated by signal 6: Aborted
db            | LOG:  aborting startup due to startup process failure
node-odm-1    | info: Authentication using NoTokenRequired
broker        | 1:C 17 Jul 03:41:53.615 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
broker        | 1:C 17 Jul 03:41:53.615 # Redis version=4.0.8, bits=64, commit=00000000, modified=0, pid=1, just started
broker        | 1:C 17 Jul 03:41:53.615 # Warning: no config file specified, using the default config. In order to specify a config file use redis-server /path/to/redis.conf
node-odm-1    | info: Listening on 0.0.0.0:6367 UDP for progress updates
broker        | 1:M 17 Jul 03:41:53.615 * Running mode=standalone, port=6379.
[... 420 lines cut ...]
worker        | psql: could not translate host name "db" to address: No address associated with hostname
worker        | Postgres is unavailable - sleeping
worker        | psql: could not translate host name "db" to address: No address associated with hostname
worker        | Postgres is unavailable - sleeping
worker        | psql: could not translate host name "db" to address: No address associated with hostname
worker        | Postgres is unavailable - sleeping
worker        | psql: could not translate host name "db" to address: No address associated with hostname
worker        | Postgres is unavailable - sleeping
worker        | psql: could not translate host name "db" to address: No address associated with hostname
worker        | Postgres is unavailable - sleeping
worker        | psql: could not translate host name "db" to address: No address associated with hostname
worker        | Postgres is unavailable - sleeping
worker        | psql: could not translate host name "db" to address: No address associated with hostname
worker        | Postgres is unavailable - sleeping
worker        | psql: could not translate host name "db" to address: No address associated with hostname
worker        | Postgres is unavailable - sleeping
worker        | psql: could not translate host name "db" to address: No address associated with hostname
worker        | Postgres is unavailable - sleeping
worker        | psql: could not translate host name "db" to address: No address associated with hostname
worker        | Postgres is unavailable - sleeping
worker        | psql: could not translate host name "db" to address: No address associated with hostname
worker        | Postgres is unavailable - sleeping
worker        | psql: could not translate host name "db" to address: No address associated with hostname
worker        | Postgres is unavailable - sleeping
worker        | psql: could not translate host name "db"

pierotofy, 2019-07-17:
Hi @silverefendy,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-07-19 23:55) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

silverefendy, 2019-07-17:
Sorry i'll try again.
How did you install WebODM? (Docker, natively, ...)?
I install WebODM from WebODM website long time ago.
Install it using ubuntu terminal. Using Docker perhaps (not really sure)

What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
http://whatismybrowser.com/w/3HHZZR8

What is the problem?
My probelm is i cannot start WebODM again.
As the error only shows 
" worker | psql: could not translate host name "db" to address: No address associated with hostname
worker | Postgres is unavailable - sleeping"
non stop

What should be the expected behavior?
It should be as always without error and show ip for webodm at the end of the terminal.

How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
Before this error, i already merge 3 dataset of photo with same configuration.
First dataset contain about 300 photos
Second dataset contain about 850 photos
Third dataset contain about 800 photos
At the fourth (around 800photos), my pc slow down and got error like not enough memory (not really remember)
Then it got very slow, i try to restart. But cannot restart, so i reset it.
After that, i try restarting webodm again 
"sudo ./webodm.sh restart"
Then that problem shows like above.
My upload speed really slow. I'll try upload it.

pierotofy, 2019-07-18:
Could we move this to https://community.opendronemap.org as it looks like a configuration / setup problem and not necessarily a bug? Thank you! :pray:

closed 2019-07-18 by pierotofy

silverefendy, 2019-07-19:
Ok,I’ll post it there again.