WebODM/WebODM #709: ./webodm.sh update Error

DykstraJosh, 2019-07-31:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Windows 10 v1903
Chrome 75 on Windows 10

### What is the problem?

When updating, I receive the attache error. Everything works, I just can't run the update. 

### What should be the expected behavior?

Update

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

[ODM_Update_Error.txt](https://github.com/OpenDroneMap/WebODM/files/3453444/ODM_Update_Error.txt)

pierotofy, 2019-07-31:
Hey @DykstraJosh :hand: give the directory a `git reset --hard` before running `./webodm.sh update`.

CMS2268, 2019-08-02:
Not sure if this is the same problem but WebODM Manager 1.8 said that it needed to be updated and gave me a "Congratulations! WebODM is downloading" notice. I have let this run for hours and it hasn't updated yet. I reinstalled 1.8 then installed 1.6.5 but nothing. Was on wifi so I plugged in the cat5 and let it run all night but no joy!  All my antivirus and firewalls are off. In the lower left of the screen is shows "idle". I already did this update on my desktop with no problem. Anyone have any ideas. Thanks for any help.

DykstraJosh, 2019-08-02:
Hey @pierotofy git reset --hard, gave me the same error

DykstraJosh, 2019-08-02:
I ended up starting from scratch.

CMS2268, 2019-08-02:
SO when you say you started from scratch you uninstall WebOBM and Docker? Did you delete all the maps?

closed 2020-04-02 by pierotofy