WebODM/WebODM #720: Construction Plan Overlay  - Would be a good option

CSParnell, 2019-08-26:
Hi team, DroneDeploy have released the following 
https://blog.dronedeploy.com/compare-design-plans-and-drone-maps-with-new-overlay-tool-161f44b7ea8a

I would love to see this feature on ODM as I feel its a far better platform.

Is this possible at all?

pierotofy, 2019-08-26:
Hi @CSParnell,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-08-28 22:10) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

CSParnell, 2019-08-27:
The ultimate goal would be that you can import a CAD file or PNG with transparent background or if another format can be imported that can have the background erased (white) then you could overlay this over the compiled TIFF. you would require the ability to manipulate the size of the import construction drawing to fit over the mapped TIFF. I am looking at it from a solar install point of view but this could be applied to any construction. 
If say a building was being constructed and you were mapping the progress over the life of the build you could overlap the construction drawings so your client could see what is where and the areas in relation to the actual mapped TIFF.

CSParnell, 2019-08-27:
Yes correct that would be ideal ability, sometimes the constructor or owner will just issues a set of drawings with no geo reference.

CSParnell, 2019-10-16:
Hi Stephen, Do you know if this has been taken up as something that will be looked into programming in?

smathermather, 2019-10-17:
So far, while there is expressed interest, there is no one who has volunteered the time or funding for this feature.

pierotofy, 2019-12-03:
@CSParnell currently you can drag & drop both Shapefiles and GeoJSON to the map; perhaps this is a sufficient workaround?

closed 2026-02-20 by pierotofy