WebODM/WebODM #722: Make permissions management easier

pierotofy, 2019-09-07:
Currently the only way to manage permissions is to use Django's complicated admin UI. This should be improved.

anwaysomani, 2019-09-29:
Hey...are you expecting a Django (HTML) template UI-based permission management system, or do you want the admin layout to be refurbished for UI...?

pierotofy, 2019-09-29:
Hey @anwaysomani :hand: ideally we'd love to have the React components in the dashboard to include a UI for handling permissions (without using the Django admin pages).

[image]

Not a quick task for sure. But we're also open to other suggestions (that perhaps require less effort)!

pierotofy, 2019-09-29:
Perhaps something as simple as a button to link the Dashboard UI to the Django admin permission page might work (or open the Django admin permission page in a popup). :+1:

anwaysomani, 2019-09-29:
Well, @pierotofy , I shall like to give it a try with React & Django. Will get back to you with an update.

pierotofy, 2021-11-01:
This has been implemented.

closed 2021-11-01 by pierotofy