WebODM/WebODM #727: WebODM fails a task if a connection error happens

pierotofy, 2019-09-11:
In some instances, a task might be flagged as "failed" although NodeODM is still processing the task (and returned an error due to a "hiccup").

This should be fixed by making WebODM more fault tolerant (retry).

A user reported the following for the task failure description:

`Unauthorized. Do you need to set a token?`

Even though NodeODM was still processing.

pierotofy, 2021-06-02:
Also happens if a process "hangs" for longer than the allowed timeout (happens if the OS starts swapping RAM), freezing the computer state for longer than ~30 seconds.

smathermather, 2025-07-17:
Agreed. I don't think I have seen it in the last 18 months, even on very long running projecfs.