WebODM/WebODM #732: Volume calculation returns Invalid response: [object Object]

uresplatform, 2019-09-20:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 76 on Windows 10

### What is the problem?

I am unable to obtain volume measurement, I keep getting the following error: Invalid response: [object Object]

No error appeared on the console, and I could not find any log files.

In the web console, the returned object is: {"error":""}

[image]

### What should be the expected behavior?

One of the two:
- volume of the selected area
- clear error message

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I have tried processing the images with both profiles: Volume measurement, High Resolution. I then selected the area for which I would like to calculate the volume.

[image]

pierotofy, 2019-09-23:
Hey @uresplatform :hand: thanks for reporting this! Does this happen with every dataset or just on some? Does selecting another area (perhaps a smaller polygon) work or do you get the same error?

pierotofy, 2019-09-23:
Should be fixed as soon as #735 is rebuilt and pushed to docker (in a few hours). :+1:

closed 2019-09-23 by pierotofy

uresplatform, 2019-09-26:
Hello @pierotofy ! Thanks for looking into this. Just updated to the latest version and works perfectly!

PR #735 (2019-09-23): Downgrade to GRASS 7.6
Fixes #732 

GRASS 7.8 broke (or fixed) something, but the output from `r.volume` is not showing in the output.

7.8 output:

```
Starting GRASS GIS...
Creating new GRASS GIS location <location8>...
Cleaning up temporary files...
Executing <sh script.sh> ...
-PROJ_INFO-------------------------------------------------
name       : WGS 84 / UTM zone 15N
datum      : wgs84
ellps      : wgs84
proj       : utm
zone       : 15
no_defs    : defined
-PROJ_EPSG-------------------------------------------------
epsg       : 32615
-PROJ_UNITS----------------------------------------- [... cut]