WebODM/WebODM #738: Could not load point cloud

capitalPSA, 2019-10-03:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? 

Chromium 77 on Ubuntu Linux

### What is the problem?

"Could not load point cloud. This task doesn't seem to have one. Try processing the task again."

### What should be the expected behavior?

Pointcloud should be viewed in the web browser / potree.

### How can we reproduce this? 

401 images were processed.
The file size of the point cloud file is 2.3 GB. 
The point cloud (.ply) file can be downloaded from WebODM and viewed in meshlab.
The WebODM / potree GUI cannot display the point cloud.
The relevant error that appears in the task output is:
```
Scanning input

1/1: odm_filterpoints/point_cloud.ply

SRS could not be determined

Encountered an error: No points found!
```

pierotofy, 2019-10-14:
Thanks for the report @capitalPSA ! This looks like a bug affecting non-georeferenced point clouds. We'll try to fix it.

jur3, 2019-10-17:
Same issue but different error in docker logs.

`WARNING Not Found: /api/projects/5/tasks/97789419-1845-4745-83ad-ca5e5d1dc3a1/assets/potree_pointcloud/cloud.js`

pierotofy, 2019-10-23:
This shoud have been fixed as part of https://github.com/OpenDroneMap/NodeODM/pull/98. Update in a few hours, then see if the problem persists?

Thanks!

closed 2019-10-23 by pierotofy