WebODM/WebODM #745: Remove glyphicon icons

pierotofy, 2019-11-07:
We should remove glyphicons in favor of font-awesome. Currently we use both.

Mingos1, 2020-12-23:
Hi there, can I work on this issue?

pierotofy, 2020-12-23:
Of course @Mingos1 :+1:

Mingos1, 2020-12-23:
Currently finishing up, what do I do with a glyphicon without a FA equivalent? 

For example, "glyphicon glyphicon-saved" does not have one.

pierotofy, 2020-12-23:
How about `fa fa-download` ?

Glyphicon and font awesome have different names, but they often have equivalent icons. Make sure to replace them by look, not by name! You can test the changes by starting WebODM with `./webodm.sh restart --dev`.

Mingos1, 2020-12-23:
Thank you, going to add when I start working in a few. 

Also, what can I do with the glyphicons integrated as class names?

pierotofy, 2020-12-23:
> Also, what can I do with the glyphicons integrated as class names?

Can you explain more details? I'm not sure I understood the question. :pray:

Mingos1, 2020-12-24:
No worries. I came across the glyphicons in the button classNames (pictured below) and I was not sure if they could be replaced as the glyphicons within I tags (e.g.,<i className="glyphicon glyphicon-saved"></i>).

Since the glyphicons are apart of bootstrap, I think it would involve removing the over styles ( e.g., btn, btn-primary) and possibly create a conflict.

Essentially, how should I approach this?

[image]

pierotofy, 2020-12-24:
I think it works fine to replace those classes (just keep the `btn`, `btn-primary`, etc.):

[image]

Have you run WebODM in development mode yet? 

`./webodm.sh start --dev`

You can see your changes in real time.

Mingos1, 2020-12-24:
Thanks for the tip; I reread the docs and downloaded Docker, which definitely made the process a lot quicker.

I finished up replacing all of the inline icons. I left all mentions of glyphicons in the stylesheets as is (in case you wanted to revert back to glyphs) and left 1 glyphicon component in (shown below) since I couldn't find a FA equivalent.

I'm going to include this note in the PR, too.

[image]

Mingos1, 2020-12-24:
#946