WebODM/WebODM #747: WebODM missing opensfm processing outputs

SwiftGust, 2019-11-11:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

 Firefox 70 on Ubuntu Linux (18.04 LTS)

### What is the problem?

I'm not sure it is NodeODM problem or WebODM problem but since I'm using WebODM so reporting issue here.

Found this issue while digging into how to get camera poses, etc... 
I manually retrieved openSfM results from docker container running NodeODM and found this discrepancy

When I download asset (all.zip) file from WebODM interface, 
WebODM interface missing openSfM result files which could be very critical info for debugging, reviewing
which were listed here https://docs.opendronemap.org/outputs.html#list-of-all-outputs

P.S. how to visualize camera poses with pointcloud?

### What should be the expected behavior?

The compressed output files should include openSfM result as documented in https://docs.opendronemap.org/outputs.html#list-of-all-outputs 
or should provide option to contain OpenSfM, images, etc...

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Process any dataset using WebODM and Download Asset -> All Assets (all.zip)
see what it contains, it is missing openSfM related outputs and images directory

pierotofy, 2019-11-13:
Hey @SwiftGust :hand: yes correctly we don't include the OpenSfM results, mostly because there's no code capable of displaying the camera information. If WebODM included the capability to display camera poses (see #439 ) we would then add logic to return the relevant OpenSfM files (`reconstruction.json` and other relevant files).

Perhaps you might be interested in helping us add the ability to show the camera poses in the WebODM UI?

SwiftGust, 2019-11-13:
Hi @pierotofy, 
Thank you for the offer but I do not know much about web codes & UIs.
Why don't we use https://github.com/mapillary/OpenSfM/blob/master/viewer/reconstruction.html?

pierotofy, 2019-11-13:
The OpenSfM viewer does not support dense point cloud visualization, but would be the starting point to integrate camera visualization in the current potree viewer.

SwiftGust, 2019-11-14:
In my humble opinion, OpenSfM viewer is a good tool to check camera poses are well extracted and
it is better having the viewer than nothing in WebODM and we think we should progressively should add viewer for dense pointcloud.

And I have a question that how can I know GCPs well-aligned? How can I assess accuracy of the full-reconstruction?

pierotofy, 2019-11-14:
For sure; we'd welcome contributions that add such functionality!

Let's move the questions over to the forum: https://community.opendronemap.org :+1:

pierotofy, 2020-04-02:
Closing in favor of #439

closed 2020-04-02 by pierotofy