WebODM/WebODM #748: Task using Lightning Network seems to not complete

felurx, 2019-11-13:
### How did you install WebODM? (Docker, natively, ...)?

I installed the Docker image.

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Firefox 70
Linux (Manjaro)

### What is the problem?

After having a task calculated using the Lightning Network, the Task will seem to run forever. The Task Output stops after `Pushes complete - joining...`. However, I did get an E-Mail stating that the task has completed, and that I can download the results using a link in the E-Mail.
Also, interestingly, after refreshing WebODM the timer of the task jumps to the same time.

### What should be the expected behavior?
The task should appear as complete in WebODM after completing it, including the possibilities to use/see generated Ortophotos/3D models/... without having to manually cancel the task and import the results.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Options: `dsm: true, depthmap-resolution: 1000, dem-resolution: 2.0, orthophoto-resolution: 2.0` (-> High Resolution preset)

Setup Lightning, Create project, Create task, choose Lightning as processing node, upload files, start task
I already tried restarting WebODM and Resyncing in the Lightning Network tab.

(Also: OpenDroneMap is such an awesome project! Thanks a ton!)

pierotofy, 2019-11-13:
Thanks for the report @felurx :+1: 

When the task gets stuck, could you run the following and copy/paste the output here?

```
docker logs $(docker ps -aq --filter name=worker)
```

Also:

```
docker logs $(docker ps -aq --filter name=webapp)
```

It might have some hints as to what went wrong.

felurx, 2019-11-13:
Thanks for the quick reply! Here you go:

[docker_log_webapp.txt](https://github.com/OpenDroneMap/WebODM/files/3842413/docker_log_webapp.txt)
[docker_log_worker.txt](https://github.com/OpenDroneMap/WebODM/files/3842414/docker_log_worker.txt)

felurx, 2019-11-13:
Oh well, I think I just realized what might be the problem: I have a _terrible_ internet connection, so WebODM wasn't able to download the results. Welp, it would be cool if WebODM could tell you that the task is complete but that it failed in downloading the results (or at least output anything at all related to the end-user). 
Is there a way to import the results into that task, or should I just delete the task and import the results into a new task?

felurx, 2019-11-13:
Oh, and should I rename/edit this issue or close it and create a new one for the actual problem?

pierotofy, 2019-11-13:
Thanks for the logs; like you say, it looks like the problem is a connection error (not necessarily on your end); either way, WebODM should be more clear about the problem instead of just hanging there. 

You can leave this issue as-is, no need to open a new one.

pierotofy, 2020-04-02:
Hey @felurx :hand: we've made some improvements since this was opened. Let me know if the problem still happens. I will close this for now.