WebODM/WebODM #749: Distortions and 2D/3D Errors (with Node_ODM) when using "--micmac" and "--dev" parameter

u5000, 2019-11-14:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox

### What is the problem?

I've tested around with the new micmac node. And i also had a few datasets running on the normal "node_odm" node. Strangely all these datasets came out extremly distoreted. When i disabled the "--micmac" parameter the problem was gone again.

in short:
- when using "--dev" and "--micmac" paramters projects on the normal node_odm are distored
- the same datasets with the same settings looks normal on the node_odm node when used without the "--micmac" parameter. 

### What should be the expected behavior?

The results of the projects running on node_odm should be identical, regardless if i'm running "--micmac" parameter or not.

pierotofy, 2019-11-14:
Hey @u5000 :hand: thanks for reporting this. It's very likely that you are seeing differences between runs because you are processing results with MicMac when using the `--micmac` flag instead of NodeODM.

Those two flags cannot affect results.

u5000, 2019-11-15:
thanks for your response. I am very positive, that what i mean is not directly related to micmac, let me explain:

- when i start webodm with the "micmac" flag i can choose between 2 differnt nodes. First the micmac node and second the normal "odm_node". This is great because this way i could always choose which node to use.

- But as i found out, if i choose the "odm_node" the results are different (distorted) from the results i get from the odm_node without the micmac flag.

- how do i know that i did not accidentally used the "micmac" node?
 - first, i am choosing the "odm_node" manually
 - when i chosse "micmac" the results are not distoreted
 - running times are different between micmac and node_odm
 - resulting files are different between both nodes

i can't explain why this is happening and i agree that the odm_node should not produce different results when webodm is run with the "micmac" flag. But apparently this is the case, for me at least.

if i needed to guess i would think that maybe webodm has an error when telling the odm_nodes the right parameter to use for the calculation (maybe there is a mixup with a micmac parameter somewhere?). Because i get similar distortions when using wrong parameters (to high depth_resolution for example) with webodm.

pierotofy, 2022-02-28:
Closing as I'm unable to reproduce. If this is still an issue, please provide some additional details (perhaps screenshots or videos?) :pray:

closed 2022-02-28 by pierotofy