WebODM/WebODM #752: Volume Calculating volumes using Volume leangth area

MontanaVDS, 2019-11-20:
****************************************

### How did you install WebODM? (Docker, natively, ...)?

[Type answer here]

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

YOUR WEB BROWSER IS:
Chrome 78 on Windows 10

✓ Your web browser is up to date.
Is JavaScript enabled? | YesHow to enable JavaScript
-- | --
YesHow to enable Cookies
YesHow to enable third-party cookies

### What is the problem?

After update Volume calculation are not calculated Distance and area are computed but volume are not being calculated.
Yesterday 11-18-19 Volumes calculated after update to webodm from WEBODM Manager 1.6 no more volumes tried on different projects all the same

### What should be the expected behavior?

Yesterday 11-18-19 Volumes calculated after update to webodm from WEBODM Manager 1.6 no more volumes tried on different projects all the same

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Not Shure
Also my Windows Docker auto updated on 11-18-19 also

dmbworld, 2019-11-20:
Same issue here since update. All existing and previously working projects no longer display volumes. Only Perimeter and Area reported.

pierotofy, 2019-11-20:
Are you getting any error messages? Could you post a screenshot of what you are seeing?

Thank you.

dmbworld, 2019-11-20:
No error messages, attached a screen dump.
[image]

MontanaVDS, 2019-11-20:
No Errors Messages.  I Get the same as above Screen Shot

pierotofy, 2019-11-20:
Thank you! I will look into it.

pierotofy, 2019-11-21:
Try to update in a few hours, then see if it's fixed (it should).

If not, please let me know.

Thank you for the report!

closed 2019-11-21 by pierotofy

dmbworld, 2019-11-21:
Thank you @pierotofy volume measurment now working.

MontanaVDS, 2019-11-21:
Working again!!
Thanks  @pierotofy

PR #753 (2019-11-21): Fixed volume measurements
Should fix #752