WebODM/WebODM #754: Possible bug: pixel size not properly populated in OpenAerialMap export

smathermather, 2019-11-26:
### What's your browser and operating system?
Chrome 78 on Windows 10

### What is the problem?

Uploads to OpenAerialMap report the default pixel size (5.0 cm) for the metadata when the calculated resolution and resolution of the dataset is higher.

### What should be the expected behavior?

The OAM exporter should report the correct resolution

### How can we reproduce this?

Process higher resolution dataset with default parameters and upload to OAM. Check metadata on upload.

pierotofy, 2019-11-26:
Would probably need to pass https://hotosm.github.io/oam-api/#api-Meta-PostMeta the information via `gsd` parameter (https://github.com/OpenDroneMap/WebODM/blob/master/plugins/openaerialmap/public/ShareButton.jsx#L73), expect I don't know if the `dronedeploy` endpoint supports it.

PR welcome!

pierotofy, 2021-11-01:
Closing as there's been no interest to implement this (a PR is still welcome though).

closed 2021-11-01 by pierotofy