WebODM/WebODM #755: MicMac working directory problem.

maciekish, 2019-11-30:
### What's your browser and operating system?
Ubuntu x64

### What is the problem?
MicMac node is running out of disk space.

How do i change where MicMac stores temporary data? --media-dir doesn't help.

pierotofy, 2019-11-30:
Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1:

closed 2019-11-30 by pierotofy

maciekish, 2019-12-07:
Conversation? This is a bug/feature request imho.

smathermather, 2019-12-09:
It's almost always better to start the conversation at community and then escalate on github. It helps get the bug report / feature requests in the correct places, and reduces the noise.