WebODM/WebODM #756: SSL Error

ryan-shea, 2019-12-01:
### What's your browser and operating system? 

Firefox 70 on macOS (Catalina)

Ubuntu 18.04 TLS on GCE

### What is the problem?

When installing a new version of WebODM on GCE, I ran into a few errors with implementing SSL.

First, the autogeneration of certificates no longer works. The command in letsencrypt-autogen.sh includes --tls-sni-01, which is deprecated here: https://community.letsencrypt.org/t/how-to-stop-using-tls-sni-01-with-certbot/83210

When using the default --ssl flag, WebODM is then inaccessible.

I generated the keys myself with Certbot and passed them through to WebODM, only to run into another port allocation problem. On restart, WebODM allocates port 80 to a default nginx install, so the rest of the startup fails.

The only solution I’ve found is to pass a different SSL_INSECURE_PORT_REDIRECT. Then, WebODM works on port 443.

### What should be the expected behavior?

SSL Autogeneration succeeds + port 80 is unallocated on manual SSL

### How can we reproduce this?

Install any WebODM version and attempt to use SSL (Certbot is the issue here)

```
Will enable SSL (Manual)
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml -f docker-compose.ssl.yml -f docker-compose.ssl-m
anual.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml -f docker-compose.ssl.yml -f d
ocker-compose.ssl-manual.yml up --scale node-odm=1
Starting db       ... done
Starting node-odm ... done
Starting broker   ... done
Starting worker   ... done
Starting webapp   ... done
ERROR: No containers to start
Creating network "webodm_default" with the default driver
Creating db ... 
Creating webodm_node-odm_1 ... 
Creating broker ... 
Creating db
Creating broker
Creating broker ... done
Creating worker ... 
Creating worker ... done
Creating webapp ... 
Creating webapp ... error
ERROR: for webapp  Cannot start service webapp: driver failed programming external connectivity on endpoint webapp (
c40d60930f1a6dd39b78d181681188803ff7a21b7d0e98cf3d3861d5cfb06147): Error starting userland proxy: listen tcp 0.0.0.0
:80: bind: address already in use
ERROR: for webapp  Cannot start service webapp: driver failed programming external connectivity on endpoint webapp (
c40d60930f1a6dd39b78d181681188803ff7a21b7d0e98cf3d3861d5cfb06147): Error starting userland proxy: listen tcp 0.0.0.0
:80: bind: address already in use
```

see https://community.opendronemap.org/t/ssl-install-problems/3210

timotay07, 2019-12-22:
I too have experienced this issue.  I removed the certbot option "--tls-sni-01-port 8000" from the letsencrypt-autogen.sh file and then rebuilt the containers (./webodm.sh rebuild).  I'm not an expert on this, but the ssl option appears to work correctly after this.

pierotofy, 2019-12-23:
That's awesome @timotay07 :+1: 

Would you be interested in opening a pull request to fix this for other WebODM users? 

https://github.com/OpenDroneMap/WebODM/edit/master/nginx/letsencrypt-autogen.sh

pierotofy, 2019-12-31:
Should be fixed in #785. Please test and see if it's fixed? If not, please re-open this issue.

Thanks!

closed 2019-12-31 by pierotofy

timotay07, 2020-01-01:
Thanks guys for merging this pull request!  I would like to contribute more, but time is tight for me at the moment.